rsta2/circle #265: Should examples use busy waits for time delays?

kevinboone, 2021-11-18:
I note that example #2 gets a time delay by busy-waiting:

for (volatile unsigned i = 1; i <= 50000000; i++)

Is this a safe thing to do? I'd be wary of doing this on a Linux RPi because it would prevent the CPU idling, and might cause overheating. Is it a safe thing to do in Circle?

I note that there is a CUserTimer class that appears to use an interrupt. I confess that I don't fully understand the code, but I'm wondering if that provides a timer that does not rely on busy-waiting?

Or perhaps I'm just missing the point?

rsta2, 2021-11-18:
This delay loop in example 2 is a leftover from a time, when Circle provided the option to run with or without MMU enabled. The sample should demonstrate the difference in performance between both cases with this simple loop. The common methods to apply delays in a program are `CTimer::MsDelay()` and `CTimer::usDelay()`. I should probably modify this sample to use one of these functions.

The class `CUserTimer` can be used, if one wants an interrupt after a given amount of microseconds on a rolling basis. An interrupt handler function gets called, when the timer elapses and one can restart it to elapse after the same or a different amount of time again.

You may have a look at [this](https://rsta2.readthedocs.io/projects/circle/en/latest/basic-system-services/time.html) for a description of the time services of Circle. This documentation is still under construction.

rsta2, 2021-11-19:
I have updated the example. The method `CTimer::SimpleMsDelay()` should be used instead of `CTimer::MsDelay()`, if there is no instance of the class `CTimer` in the system.

rsta2, 2021-12-02:
The mentioned modification is part of the new release Circle 44.3, so this can be closed.

closed 2021-12-02 by rsta2