rsta2/circle #266: System does not boot when instantiating a large(ish) array

tonoso, 2021-11-21:
I am instantiating an object of a new class, which I declare within the struct of the miniorgan Sample. I observe that if the buffers declared in this new object are large (48000), the system doesn't boot, and the LED blinks once. When I reduce the buffer size to a small value (eg 400), the system boots correctly. I would appreciate any help on this. Thanks in advance.

rsta2, 2021-11-22:
Most likely the problem is, that this class with the large buffers is instantiated on the stack, which is only 128 KBytes in size. You should not instantiate the class as a member itself, but put a pointer into the class definition and use `new` to create the object. Then the object will be on the heap, which is much larger.

tonoso, 2021-11-23:
Thanks for this. After instantiating the object using 'new' it works fine. Consider this closed.

rsta2, 2021-11-24:
You are welcome. Good that it works now. OK, I close it.

closed 2021-11-24 by rsta2