rsta2/circle #267: Add support for I2S on GPIO 28-31 for use with RPi V1 Model B P5 header

diyelectromusic, 2021-11-21:
It would be really neat to be able to add support for the I2S interface on the original Raspberry Pi V1 Model B.  This is available via GPIO 28-31 via the "ALT 2" alternative functions on the extra P5 header near the original 26-way header.

I've added it to my own environment by adding a conditional compilation option (based on a new setting in circle/include/sysconfig.h) for i2ssoundbasedevice in the function CI2SSoundBaseDevice::RunI2S to use GPIO 28-31 and ALT 2 rather than GPIO 18-21 and ALT 0.  This works for me, but I wonder if it would be better to add a cmdline option to make it a boot-time configuration setting.

I've not really looked into the boot-time configuration options, otherwise I'd have a go myself.  If this is something you'd be interested in adding that would be great.  If it is something you are interested in, but unlikely to get around to then let me know and I'll see if I can create a PR to do it myself (although I'm still a bit of a GitHub newbie at the moment, but I'm happy to have a go).

Many thanks,
Kevin

rsta2, 2021-11-22:
No problem, I will implement your suggestion this week. I think, we do not need a configuration option here, but simply can request the board model from `CMachineInfo` and use the GPIO pins 28-31 with ALT2, when one of these models is returned:

* MachineModelA
* MachineModelBRelease1MB256

Question: Do these boards have the P5 header too?

* MachineModelBRelease2MB256
* MachineModelBRelease2MB512

diyelectromusic, 2021-11-22:
Ah, even better! Fantastic :)

I'm trying to decipher the Raspberry Pi revisions(!) I think it is the V1 Model B revision 2 that includes the P5 header.  The V1 Model B revision 1 doesn't have it, so has no I2S at all as far as I can see; the V1 Model B+ onwards has the 40-pin header, so will use the 18-21 mapping on the main GPIO header.  As far as I can see, any V1 "revision 2" based board has the P5 header - so that includes the original model A too I think (the A+ moved to the smaller board and 40-pin header I believe).  I'm largely going on the graphics for the connectors on the wikipedia page (https://en.wikipedia.org/wiki/Raspberry_Pi), but also from the RPI hardware history page (https://elinux.org/RPi_HardwareHistory).

I'm not quite sure what that means in terms of your circle definitions, but I /think/ it might apply to anything between 0x04 and 0x0F, that would mean the following are all P5 boards that could support I2S:
* MachineModelBRelease2MB256
* MachineModelA
* MachineModelBRelease2MB512

It looks like it won't apply to MachineModelBRelease1MB256 as that is the V1 model B revision 1 board with no P5 header so no I2S capability at all if I'm interpreting everything correctly!

Many Thanks,
Kevin

rsta2, 2021-11-22:
If the revisions 0x04 to 0x0F have the P5 header, the three model values, you have noted, are the right one. Thanks for figuring this out. I already applied the needed modifications to the *develop* branch. Can you please test it? Thanks.

Rene

diyelectromusic, 2021-11-22:
Yep - that seems to work.  Building 29-miniorgan with I2S enabled for a circle configuration where RASPPI = 1 means I have a kernel.img that works with my PCM5102 over I2S both with a 40-way GPIO Raspberry Pi Zero V1 (non-wireless) and with a 26-way GPIO + P5 Raspberry Pi V1 model B revision 2.

Brilliant! Many thanks :)

rsta2, 2021-11-23:
Great! Thank you for testing.

diyelectromusic, 2021-11-23:
For info, also seems to work with RASPPI = 2 on my V3 Pi. I was going to try on a V2 Pi, but I think mine might be dead!

rsta2, 2021-11-23:
Thanks for info. I tested sample/29-miniorgan with I2S on a RPi 2B again and it worked. So maybe there is really a problem with your hardware. :(

rsta2, 2021-12-02:
The I2S support for early Raspberry Pi models on the P5 header is part of the new release Circle 44.3, so this can be closed.

closed 2021-12-02 by rsta2