rsta2/circle #268: debug framebuffer virtual offset in QEMU?

incanus, 2021-11-24:
Virtual offset / double buffering does not seem to work in QEMU. Works fine on hardware, of course. I'm wondering if anyone has tips or has had success getting this to work so as to not have to debug on hardware. Only the initial buffer is shown when trying this — neither `SetVirtualOffset()` nor `WaitForVerticalSync()` seem to do anything. 

I have tested this by way of the 41st demo (2D graphics) which flickers as it is actually single-buffered, then with a reduced example that just alternates colors to clear the screen, but only the first color is ever seen. This occurs with both normal QEMU display as well as VNC to the local host.

rsta2, 2021-11-25:
I guess, QEMU simply doesn't support this. Can you disable the double-buffering for your tests with QEMU and enable it, if you build for the real hardware?

incanus, 2021-11-25:
It doesn't hurt anything in QEMU, I just can't test all of my functionality. What I am working with currently is a slightly modified `Screen` that is backed by a triple-height framebuffer. The first, top page behaves as a normal `Console` with all of the functionality from `Screen`. Then there are two more pages which can be shown/drawn via `SetVirtualOffset()` and a buffer array offset similar to how double buffering works in `2DGraphics`. All is well on device, but QEMU is never able to show or draw to the 2nd & 3rd screens of the framebuffer. 

In the case of example 41, there is no `Screen`, only `2DGraphics`, and the first page shows just fine, but as I mentioned above, the animation flickers since the 2nd page and thus double buffering are never working. 

So QEMU in all cases seems to be limited to only only framebuffer page, regardless of drawing type.

rsta2, 2021-11-26:
I understand. I'm afraid you need the real hardware to test this.

incanus, 2021-11-26:
No, worries, I had feared that but also wanted to make sure I wasn’t missing anything, as well as get this documented in this project for future reference. Thank you so much for this project and all of your work and help!

closed 2021-11-26 by incanus

rsta2, 2021-11-26:
You are welcome. Thank you for documenting this. I'm glad that you like this project.