rsta2/circle #269: PWM audio on CM4?

incanus, 2022-01-04:
I've been trying unsuccessfully to get PWM audio working on the CM4. The same hardware works fine on HDMI, as well as with PWM when booted into linux if `dtparam=audremap` is used. The same software works on a normal Pi 4B headphone jack. 

I have gotten into some bits of `machineinfo.cpp` and `pwmsoundbasedevice.cpp` to ensure that GPIO 12 & 13 are being used in all cases (regardless of not being a Pi Zero). 

Is there something else here that is either incompatible or needs to manual massaging?

rsta2, 2022-01-04:
I do not see a reason, why it shouldn't be possible to get the PWM sound working via  GPIO12/13 on the CM4. Additionally to the modifications, which are needed to use GPIO12/13 on the RPi Zero, and which you have probably already applied for the CM4, it is important to modify the code to use the standard PWM device, instead of PWM1, which is normally used with `RASPPI = 4`. This is selected [here](https://github.com/rsta2/circle/blob/master/lib/pwmsoundbasedevice.cpp#L39). You have to change this `#if` to be able to use the class `CPWMSoundBaseDevice` on the CM4 via GPIO12/13. If this doesn't help yet, please set the `CLOCK_RATE` to 125000000, as it is normally used on the RPi 4B.

incanus, 2022-01-05:
Thank you @rsta2 as ever — I had missed, and it required, both of those changes. Using the standard PWM device as well as 125000000 works great!

closed 2022-01-05 by incanus