rsta2/circle #272: C2DGraphics::DrawImage member functions

CasaDeRobison, 2022-01-18:
There are four functions in the DrawImage family: DrawImage, DrawImageTransparent, DrawImageRect, DrawImageRectTransparent.

DrawImage & DrawImageTransparent assume the source PixelBuffer is the same size as the destination. That seems reasonable.

DrawImageRect also assumes PixelBuffer is the same size as the destination, but DrawImageRectTransparent takes the dimensions of the source PixelBuffer as arguments.

It would be nice if an overload of DrawImageRect also took the source PixelBuffer dimensions as arguments.

If you are agreeable to that, I would be happy to add a member function overload and do a pull request. However, it is probably just as easy for you to do it vs reviewing a PR.

Let me know if you are interested in that feature and would like a PR. Thanks!

rsta2, 2022-01-18:
Yes, please send a PR and branch off from the *develop* branch. The class `C2DGraphics` has not been implemented by myself, so I'm not that familiar with the ideas behind, and if you can provide an additional tested version of `DrawImageRect()`, which also takes the source PixelBuffer dimensions as arguments, that would be less effort for me. Thanks!

rsta2, 2022-01-29:
I added another variant of `DrawImageRect()` to the class `C2DGraphics`. Can you please check, if it meets your requirements? The modification is on the *develop* branch, where development takes place.

rsta2, 2022-02-20:
@CasaDeRobison This issue will be closed soon and I will revert the commit 98c9e12, because I'm not very convinced, that it is of much use, as you seem to do either, because you do not respond.

rsta2, 2022-02-23:
The commit 98c9e12 has been reverted.

closed 2022-02-23 by rsta2

CasaDeRobison, 2022-03-21:
Sorry for the radio silence. My day job bit me hard. I'll actually propose something at some point when time is available again.