rsta2/circle #273: usb mouse not working

stephaneweg, 2022-01-22:
i am trying the sample 10 on the raspberry pi 4 model B

it detects when the mouse is pluged, but it does not moves the cursor when i move the mouse

i have the same problem with the sample in the ugui add on

i think there is a problem in the usbmouse driver ?

rsta2, 2022-01-23:
Can you please try *sample/10-usbmouse* again and remove [this line](https://github.com/rsta2/circle/blob/master/sample/10-usbmouse/kernel.cpp#L163) before from the sample, so that you can see, if there are any error messages from the USB driver, when the mouse is attached.

stephaneweg, 2022-01-26:
I get theses messages

usbdev1-1: Device ven3f0-1198 found
usbdev1-1: Product: HP HP USB 1000dpi Laser Mouse
usbdev1-1: Interface int3-1-2 found
usbdev1-1: Using device/interface int3-1-2
usbdev1-1: Interface int3-1-1 found
usbdev1-1: Using device/interface int3-1-1
**xhciep: : Transfer error 6 on endpoint 1**
usbhub: Port 1:  Device configured
kernel : USB mouse has 5 buttons
kernel : USB mouse has a wheel

rsta2, 2022-01-26:
Thanks for testing! The "Transfer error 6" shows a STALL condition on an USB device. The device will not functioning after that. It is a bit uncommon, that an USB mouse provides the interface `int3-1-1`, which is reserved for USB keyboards. Maybe there is a problem, because Circle will load both the mouse and the keyboard driver for this device. You can prevent this by creating a file *cmdline.txt* on the SD card with this contents in the first line:

```
usbignore=int3-1-1
```

There is some chance, that the mouse function will work with that.

stephaneweg, 2022-01-26:
Thanks, it works now, you can mark the issue as resolved (but maybe document it somewhere)

rsta2, 2022-01-27:
Thanks for info! I added the mouse with the workaround info to the [Wiki](https://github.com/rsta2/circle/wiki/Device-compatibility-list).

closed 2022-01-27 by rsta2