rsta2/circle #274: Idea: MiniDexed, a FM synthesizer on Circle

probonopd, 2022-01-31:
Dexed is a multi platform, multi format synth that is closely modeled on the famous DX7 synthesizer by a well-known Japanese manufacturer.
While https://github.com/rsta2/minisynth/ implements subtractive "analog" synthesis, while Dexed implements additive "digital" FM synthesis. So it would be a great complement.

I am wondering whether Dexed could be ported to Circle, in order to recreate basically an open source equivalent of the TX802 (8 DX7 instances without the keyboard in one box).

Dexed has a sound engine written in C++ and has already been ported to:
* Teensy Microcontroller using the Arduino environment: [MicroDexed](https://codeberg.org/dcoredump/MicroDexed)
* lvtk (LV2): [dexed.lv2](https://github.com/dcoredump/dexed.lv2)

The author of MicroDexed says he'd be happy to see a it on Raspi-bare-metal and while he doesn't have the time to do the port on his own, he is happy to help by answering questions. I hope a lot could be reused from https://github.com/rsta2/minisynth/, but the sound engine would have be to swapped out. 

Performance wise, a Raspberry Pi 4 1GB running REAPER on Linux could handle 8 (and probably more) instances of Dexed simultaneously in my tests.
And it looks like a lot of code, e.g., for handling USB MIDI keyboards, etc. is already there in minisynth.

rsta2, 2022-02-01:
This is an interesting project idea and the Raspberry Pi 4 (maybe 3 too) should have the required performance. I would also support such a project by answering questions regarding Circle or MiniSynth Pi.

nikitalita, 2022-02-02:
An LV2 host on circle would be ideal, because it would enable many other synths to be easily ported to Circle as well. However, I'm not certain if the effort involved would be huge or not. Looking at the libraries available, https://github.com/lv2/lilv seems to be the only game in town, and nearly all current host implementations use it. I notice that there are a lot of OS-specific compiler pragmas that might make this a pain to port to Circle. 
There WAS a LV2 host that was written for a baremetal Raspberry Pi (located here: https://github.com/Joeboy/pixperiments/tree/master/pitracker) but it appears to be experimental and unfinished.

dcoredump, 2022-02-02:
> Dexed is a multi platform, multi format synth that is closely modeled on the famous DX7 synthesizer by a well-known Japanese manufacturer. While https://github.com/rsta2/minisynth/ implements subtractive "analog" synthesis, Dexed implements additive "digital" FM synthesis. So it would be a great complement.
> 
> I am wondering whether Dexed could be ported to Circle, in order to recreate basically an open source equivalent of the TX802 (8 DX7 instances without the keyboard in one box).
> 
> Dexed has a sound engine written in C++ and has already been ported to:
> 
>     * Teensy Microcontroller using the Arduino environment: [MicroDexed](https://codeberg.org/dcoredump/MicroDexed)
> 
>     * lvtk (LV2): [dexed.lv2](https://github.com/dcoredump/dexed.lv2)
> 
> 
> The author of MicroDexed says he'd be happy to see a it on Raspi-bare-metal and while he doesn't have the time to do the port on his own, he is happy to help by answering questions. I hope a lot could be reused from https://github.com/rsta2/minisynth/, but the sound engine would have be to swapped out.
> 
> Performance wise, a Raspberry Pi 4 1GB running REAPER on Linux could handle 8 (and probably more) instances of Dexed simultaneously in my tests. And it looks like a lot of code, e.g., for handling USB MIDI keyboards, etc. is already there in minisynth.

Hi! I am the one who used the Dexed engine and put it into LV2. This one we are using on Zynthian (https://zynthian.org) - a Raspi based multi purpose Synth. The same engine I am using for MicroDexed (based on a Teensy (3.6/4.1).

The engine for the Teensy is located [here](https://codeberg.org/dcoredump/Synth_Dexed). This could be the base for a usage based on circle. The central rendering method is void `AudioSynthDexed::update(void)` and inside this method the function `getSamples(AUDIO_BLOCK_SAMPLES, lblock->data);` is the _magic_ one which generates `AUDIO_BLOCK_SAMPLES` and puts the samples in an array `lblock->data` (an array of the length `AUDIO_BLOCK_SAMPLES`).

Unfortunately I don't have enough time to work on this project permanently, but I can help to find the right places in the code. It can't be very complicated. Don't hrsitate to contact me by email (dcoredump@googlemail.com).

Regards, Holger

probonopd, 2022-02-02:
> An LV2 host on circle would be ideal, because it would enable many other synths to be easily ported to Circle as well. 
> (...)
> There WAS a LV2 host that was written for a baremetal Raspberry Pi (located here: https://github.com/Joeboy/pixperiments/tree/master/pitracker)

> I am the one who used the Dexed engine and put it into LV2 (...) The same engine I am using for MicroDexed

What do the experts think about the use of LV2 on Circle? Useful abstraction or unnecessary overhead hassle?

Fwiw, just stumbled upon https://github.com/dwhinham/mt32-pi: a bare metal MIDI synthesizer for the Raspberry Pi 3 or above, based on the Munt Roland MT-32 emulator, FluidSynth SoundFont player, and Circle.

dcoredump, 2022-02-03:
> What do the experts think about the use of LV2 on Circle? Useful abstraction or unnecessary overhead hassle?

I won't call me an expert. On one hand it sounds really cool, but on the other hand there wouldn't be much LV2 clients which would work (IMHO). Most LV2 plugins have much dependencies to other libraries and you have to port them also to circle. Most plugins have GUIs for X11... I think you have to strip down everything. Only the raw sound engine has to be there - and the LV2 code. That's exactly what I have done with Dexed. If you have to write the LV2 code around a stripped down synth, it would be easier to use the synth code directly with circle.

But as I said: I am not an expert - maybe there are better ways to do this...

Regards, Holger

rsta2, 2022-02-03:
I think, one of the strengths of Circle is, that you don't have that much software layers to deal with, which is good for understanding and performance. That's why to keep it simple I would vote for a solution without LV2 too, at least for the moment.

I tried to build the code in the [Source/msfa/ subdirectory](https://github.com/asb2m10/dexed/tree/master/Source/msfa) of Dexed with Circle and despite from two warnings, I wasn't able to fix so far, that worked. I think, this directory contains the basic synth engine of Dexed. I used the [Standard C and C++ Library Support for Circle](https://github.com/smuehlst/circle-stdlib) project, not Circle directly, because the msfa code requires Standard C and C++ bindings.

dcoredump, 2022-02-03:
Nice @rsta2!

The msfa directory is the main engine for Dexed but around this engine there are some improvements/extensions. I also manually added a portamento patch for the Teensy object Synth_Dexed. Furthermore there is a loader/extractor for sysex data, so you can load sysex sounds directly. Perhaps I can find time to try to add the Sysnth_Dexed code into the mentioned C/C++ project at weekend.

Regards, Holger

probonopd, 2022-02-03:
> I tried to build the code in the [Source/msfa/ subdirectory](https://github.com/asb2m10/dexed/tree/master/Source/msfa) of Dexed with Circle and despite from two warnings, I wasn't able to fix so far, that worked.

Great! Would you mind uploading your almost-working code somewhere? That way we could all have a look to see if we can find out a way to fix the two warnings.

rsta2, 2022-02-03:
I meant, I was able to build the msfa code, it did not run yet. The files, which I used to build msfa in the circle-stdlib environment, are in the attached archive. You have to clone the [circle-stdlib](https://github.com/smuehlst/circle-stdlib) project with submodules libs/circle and libs/circle-newlib and need one of the toolchains, mentioned in the README.md file. Then the path to the circle-stdlib root directory must be set in the Makefile, which is provided in the archive below. The circle-stdlib project must be configured and build separately for the target Raspberry Pi model. Then you can go to src/ directory in the archive and do make. Let me know, when you have questions.

[src.zip](https://github.com/rsta2/circle/files/7998130/src.zip)

probonopd, 2022-02-05:
Thanks @rsta2. So here is what I did:

```
uname -a
# FreeBSD 13.0-RELEASE-p4
sudo pkg install devel/gcc-arm-embedded
export PATH=/usr/local/gcc-arm-embedded-10-2020-q4-major/bin/:$PATH
mkdir MiniDexed
cd MiniDexed
git clone --recursive https://github.com/smuehlst/circle-stdlib
wget "https://github.com/rsta2/circle/files/7998130/src.zip"
cd src/
gmake
# gmake: *** No rule to make target 'iostream', needed by 'kernel.o'.  Stop.
```

I assume this is because I have not build `circle-stdlib` yet. So I do:

```
cd ../circle-stdlib
./configure -r 4 # Raspberry Pi 4
# (...)
# configure: creating ./config.status
# config.status: creating Makefile

gmake
# cd libs/circle && ./makeall --nosample
# make[1]: "/usr/home/user/MiniDexed/circle-stdlib/libs/circle/tools/Makefile" line 5: Need an operator
# (...)
# make[1]: Fatal errors encountered -- cannot continue
# make[1]: stopped in /usr/home/user/MiniDexed/circle-stdlib/libs/circle/tools
# gmake: *** [Makefile:10: circle] Error 1
```

From there I ran into FreeBSD related build issues for which I will open another ticket. So, next attempt should be to build this on Linux.

rsta2, 2022-02-05:
FreeBSD is not supported as build platform, only Linux and Windows (to some degree).  You definitely need a toolchain from [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads), the C++ standard library support will not work otherwise.

probonopd, 2022-02-05:
Started a repository over at https://github.com/probonopd/MiniDexed/. First step is to get it to compile on GitHub Actions CI.

probonopd, 2022-02-05:
There are now continuous CI builds:
https://github.com/probonopd/MiniDexed/releases/tag/continuous

I'd be thrilled if we could find a way to get any sound out of it.

probonopd, 2022-02-06:
@rsta2 looking at https://github.com/rsta2/circle/blob/master/sample/29-miniorgan/miniorgan.cpp, it seems like it contains the logic to get MIDI data, convert it to sound samples, and feed the sound data into the sound system using [`unsigned CMiniOrgan::GetChunk (u32 *pBuffer, unsigned nChunkSize)`](https://github.com/rsta2/circle/blob/75bc0890490f4c38f8cdf0eee83a2bad40942566/sample/29-miniorgan/miniorgan.cpp#L208). Is that correct? Should we start from using `miniorgan.cpp` to send the MIDI packets to Dexed, retrieve the sound samples from Dexed and send them to the sound system? 

@dcoredump do you know how one sends MIDI packets to the Dexed engine, and how one gets sound data back from Dexed?

rsta2, 2022-02-06:
@probonopd I think, the sample/29-miniorgan is a good base to start with Circle sound. It's correct, `GetChunk()` gets called, when new sound samples need to be placed into the given buffer.

Please have a look into [doc/bootloader.txt](https://github.com/rsta2/circle/blob/master/doc/bootloader.txt) for using a serial bootloader, which I do for development.

probonopd, 2022-02-06:
Trying to port over sample/29-miniorgan to https://github.com/probonopd/MiniDexed I am running into errors like

```
kernel.cpp: In constructor 'CKernel::CKernel()':
kernel.cpp:8:31: error: no matching function for call to 'CScreenDevice::CScreenDevice()'
    8 | : CStdlibAppStdio ("minidexed")
      |                               ^
In file included from ../circle-stdlib/include/circle_stdlib_app.h:20,
                 from kernel.h:7,
                 from kernel.cpp:4:
../circle-stdlib/libs/circle/include/circle/screen.h:158:2: note: candidate: 'CScreenDevice::CScreenDevice(unsigned int, unsigned int, boolean, unsigned int)'
```

Is it my cursory understanding of C++ or are there differences between Circle and circle_stdlib that make this more than a copy-and-paste exercise?

smuehlst, 2022-02-06:
Without the concrete source code this is hard to tell. I see that the minimal test program in https://github.com/probonopd/MiniDexed/tree/main/src builds without error. How is your new program different?

probonopd, 2022-02-06:
@smuehlst thanks for looking into this. I have pushed the code that doesn't build to the [`miniorgan`](https://github.com/probonopd/MiniDexed/tree/miniorgan) branch. It's probably just me doing something stupid.

smuehlst, 2022-02-06:
I created a pull request in the MiniDexed repository with a fix for the build errors. You are right that it is not a simple copy-and-paste exercise because the circle-stdlib convenience base class CStdlibAppStdio() already bundles most of the individual classes that are used explicitly in the Circle miniorgan sample program. So several things that happen explicitly in the Circle miniorgan program happen implicitly in the circle-stdlib variant.

probonopd, 2022-02-06:
Excellent, thank you very much for you help @smuehlst. Now we should have MIDI input working (I will test this soon) and the Dexed code building. Next we need to figure out how to use those MIDI messages to feed them into Dexed, and how to get the sound data out of Dexed and feed it into the circle-stdlib sound system.

probonopd, 2022-02-06:
Oops! Now it stalls on the "rainbow" Raspberry Pi boot screen, does not blink 5 times.

rsta2, 2022-02-06:
Please apply the following patch and it should work as before:

```
diff --git a/src/miniorgan.cpp b/src/miniorgan.cpp
index 25ad6e0..df1a03b 100644
--- a/src/miniorgan.cpp
+++ b/src/miniorgan.cpp
@@ -76,7 +76,7 @@ CMiniOrgan::CMiniOrgan (CInterruptSystem *pInterrupt, CI2CMaster *pI2CMaster)
 	),
 	m_pMIDIDevice (0),
 	m_pKeyboard (0),
-	m_Serial (pInterrupt, TRUE),
+	m_Serial (pInterrupt, TRUE, 5),
 	m_bUseSerial (FALSE),
 	m_nSerialState (0),
 	m_nSampleCount (0),
```

The problem was, that the serial interface at GPIO14/15 is already initialized in the constructor of the class `CStdlibAppStdio` to be used to display the log for example. In `CMiniOrgan::CMiniOrgan()` there was another initialization for the same interface, which triggers an assertion failed. Unfortunately this assertion cannot be shown that early, because the screen initialization is not completed at that stage.

The solution is to use an other serial interface at GPIO12/13 in `CMiniOrgan` for serial MIDI. This only works on the Raspberry Pi 4, which has multiple serial interfaces.

I don't know, if you use the serial bootloader in the meantime. If so, please use the new bootloader (Flashy) for that large kernel images, as described in *doc/bootloader.txt*. It loads quicker and the older bootloader sometimes has problems with larger kernel images.

probonopd, 2022-02-06:
Thanks @rsta2, now I no longer get the crash. But I see "Function is not supported" under the entry for my USB MIDI keyboard, and I can hear no sound when I play (the same hardware works with https://github.com/rsta2/minisynth/). Can you spot what I am missing? Sorry for asking so many questions!

rsta2, 2022-02-06:
The miniorgan wasn't initialized and started properly yet. Another patch:

```
diff --git a/src/kernel.cpp b/src/kernel.cpp
index dcc6bf9..d0eb3ab 100644
--- a/src/kernel.cpp
+++ b/src/kernel.cpp
@@ -18,12 +18,31 @@ CKernel::~CKernel(void)
 
 bool CKernel::Initialize (void)
 {
-	return CStdlibAppStdio::Initialize ();
+	if (!CStdlibAppStdio::Initialize ())
+	{
+		return FALSE;
+	}
+
+	if (!m_MiniOrgan.Initialize ())
+	{
+		return FALSE;
+	}
+
+	return TRUE;
 }
 
 CStdlibApp::TShutdownMode CKernel::Run (void)
 {
 	std::cout << "Hello MiniDexed!\n";
 
+	m_MiniOrgan.Start ();
+
+	while (m_MiniOrgan.IsActive ())
+	{
+		boolean bUpdated = mUSBHCI.UpdatePlugAndPlay ();
+
+		m_MiniOrgan.Process (bUpdated);
+	}
+
 	return ShutdownHalt;
 }
```

Some MIDI keyboards do have additional USB interfaces, which are not needed for the normal MIDI function and which are not supported by Circle. Normally you can ignore this "Function is not supported" message here. If your keyboard works with MiniSynth, it should work now with the upcoming MiniDexed too. ;)

Yes, you can feed serial MIDI to GPIO13. This is a SoC number, not the position on the header (see [this](https://pinout.xyz/pinout/uart)).

BTW. There is also a bootloader, which works over Ethernet using TFTP or HTTP. You have to build the Circle sample/38-bootloader for this.

smuehlst, 2022-02-06:
@rsta2 I see that the `CMiniOrgan` class has a member of type `CSerialDevice`. The class `CStdlibAppScreen` that is used indirectly via base class `CStdlibAppStdio` also does have a member of type `CSerialDevice`. Can that be a problem? I remember from the past that certain device classes must only be instantiated once in an application.

probonopd, 2022-02-06:
Thanks, now I hear sound when I press keys on the USB MIDI keyboard! :+1:  and also when I press keys on the USB computer keyboard.

One strange thing though: When I use the computer keyboard, then I can play long notes, but when I use the MIDI keyboard which is attached to the original raspberry Pi keyboard which is attached to a Raspberry Pi 4 **USB 3** port. then they are all "staccato" independently of how long I keep a key pressed. It works when I use the MIDI keyboard which is attached to the original raspberry Pi keyboard when it is attached to a Raspberry Pi 4 **USB 2** port.

probonopd, 2022-02-06:
Now the :100: question is, how can we get Dexed (which already is getting built as part of [https://github.com/probonopd/MiniDexed](https://github.com/probonopd/MiniDexed/tree/main/src)) or Synth_Dexed (which would need to be ported first) hooked in? Perhaps @dcoredump can have a look or give some hints?

rsta2, 2022-02-06:
@smuehlst Yes, there was only one instance of `CSerialDevice` allowed in the past, but this changed with the Raspberry Pi 4, which has multiple instances of the UART. My first patch changed the instance of the serial device in `CMiniOrgan` to device number 5, according to [this mapping](https://github.com/rsta2/circle/blob/master/include/circle/serial.h#L33). After this both instances of `CSerialDevice` work with different hardware devices.

rsta2, 2022-02-06:
@probonopd When I test this with my AKAI LPK25 keyboard, there are no problems on both USB2 and USB3 sockets. The tone sounds as long a key is pressed without interruption. Even when I insert an USB2 hub between the USB3 socket and the LPK25, it does work. Unfortunately I have currently no explanation for this behavior with your hardware.

dcoredump, 2022-02-06:
> Now the :100: question is, how can we get Dexed (which already is getting built as part of [MiniDexed](https://github.com/probonopd/MiniDexed/tree/main/src)) or Synth_Dexed (which would need to be ported first) hooked in? Perhaps @dcoredump can have a look or give some hints?

Nice work, guys!

I am currently very short of time. I hope I can write some hints tomorrow.

dcoredump, 2022-02-07:
So, let's go! First, thanks to all involved for the way here - I'm looking forward to a Raspi-Dexed :)

First, an object `AudioSynthDexed(uint8_t max_notes, uint16_t sample_rate);` must be created initially.

`dexed=new AudioSynthDexed(16,44100);`

Then the object must be activated:

`dexed.activate();`

After that you can then filter on the MIDI channel in the MIDI loop and as soon as a NoteOn or Note-Off message arrives, the corresponding key on the dexed can be virtually pressed/released:

```
void keydown(int16_t pitch, uint8_t velo);
void keyup(int16_t pitch);
void setSustain(bool sustain);
```

In the audio interrupt the corresponding audio data would have to be calculated. This is done by

`     void getSamples(uint16_t n_samples, int16_t* buffer); `

where n_samples is the number of samples to be calculated (for Teensy 128, depending on the buffer size the Raspi processes 64, 128, 256, ...?). The higher the value, the higher the latency. 128 bytes corresponds to a latency of 2.9ms. `int16_t* buffer` is the pointer to the buffer, where the data comes in.

for example:

```
...
int16_t audio_buf[128];
...
dexed.getSamples(128,audio_buf);
...
```

After that the `audio_buf` data can be processed to the audio output or before that several voices can be added by a software mixer.

So far... I hope this helps for the beginning.

Regards, Holger

dcoredump, 2022-02-07:
If this (-^) works: Perhaps I should add the MIDI-handling code from MicroDexed into AudioSynthDexed, so that it is not necessary to maintain the code for it in several places.

probonopd, 2022-02-07:
Thanks @dcoredump. Very helpful for getting started. Since you are talking about `AudioSynthDexed` I assume your recommendation is to use https://codeberg.org/dcoredump/Synth_Dexed/src/branch/master/src instead of https://github.com/asb2m10/dexed/tree/master/Source/msfa (which is the one @rsta2 [got to build on circle-stdlib](https://github.com/probonopd/MiniDexed/tree/main/src/msfa))? Wouldn't such a port be significant work because it contains Arduino-specific concepts?

In any case, I tried to start porting Synth_Dexed (in the [`synth_dexed` branch](https://github.com/probonopd/MiniDexed/tree/synth_dexed)) but compilation fails with

```
$ make
  CPP   miniorgan.o
In file included from miniorgan.cpp:25:
synth_dexed.h:1304:43: error: expected class-name before ',' token
 1304 | class AudioSynthDexed : public AudioStream, public Dexed
      |                                           ^
synth_dexed.h:1311:67: error: 'AUDIO_BLOCK_SAMPLES' was not declared in this scope
 1311 |     const uint16_t audio_block_time_us = 1000000 / (SAMPLE_RATE / AUDIO_BLOCK_SAMPLES);
      |                                                                   ^~~~~~~~~~~~~~~~~~~
synth_dexed.h: In constructor 'AudioSynthDexed::AudioSynthDexed(uint8_t, uint16_t)':
synth_dexed.h:1308:64: error: class 'AudioSynthDexed' does not have any field named 'AudioStream'
 1308 |     AudioSynthDexed(uint8_t max_notes, uint16_t sample_rate) : AudioStream(0, NULL), Dexed(max_notes,sample_rate) { };
      |                                                                ^~~~~~~~~~~
make: *** [../circle-stdlib/libs/circle/Rules.mk:143: miniorgan.o] Error 1
```

It looks like [AudioStream](https://github.com/PaulStoffregen/cores/blob/master/teensy3/AudioStream.h) is a Teensy specific concept. How would one replace or port this to circle-stdlib?

dcoredump, 2022-02-07:
I would prefer to use something like AudioSynthDexed because there already are several additions for using Dexed on a microcontroller.

Yes, currently it uses the Teensy audio stack which generates thecerror messages. I would try to build a better version which uses the Teensy specific parts only when enabled.

Give me a day for those additions.

BTW: Has circle something like a audio subsystem (or a audio interrupt function) and what is the buffer size?

probonopd, 2022-02-07:
> Give me a day for those additions.

Very cool, this is tremendous! I could go nowhere without your help here.

> BTW: Has circle something like a audio subsystem (or a audio interrupt function) and what is the buffer size?

@rsta2 wrote above,

> It's correct, GetChunk() gets called, when new sound samples need to be placed into the given buffer.

Maybe this already helps? Otherwise we need to ask @rsta2 to elaborate on how exactly to "feed" the "audio subsystem".

probonopd, 2022-02-07:
> When I test this with my AKAI LPK25 keyboard, there are no problems on both USB2 and USB3 sockets. 

Let's treat it as a quirk of my hardware for now and not worry about it.

probonopd, 2022-02-07:
Ideally our solution will work with all 3 audio outputs mentioned, but if we need to pick one to start with it would be the 3.5" headphone jack because that's the hardware _every_ Raspberry Pi has.

@rsta2 thought about USB audio class devices?

rsta2, 2022-02-07:
> are you considering to support USB audio class devices at some point?

Currently not. The USB driver does not support isochronous transfers, which are required for USB audio devices, and it is difficult to implement this on the Raspberry Pi.

dcoredump, 2022-02-08:
Thanks @rsta2 for the programming hints!

> > Give me a day for those additions.
> 
> Very cool, this is tremendous! I could go nowhere without your help here.

Ok, I was probably a bit hasty there. This will take a little longer, because I first have to learn how to work with the sound backend of circle and therefore some adjustments are necessary.

I got some things right, but there are still some bigger problems, where my C++ knowledge seems to be insufficient (constructor of the class `AudioSynthDexed`, which inherits from `Dexed` and `CI2SSoundBaseDevice` and a replacement for the `millis()` object from the Teensy classes).

You can include the whole thing by deleting the `msfa` directory and cloning `SynthDexed`:

```
cd MiniDexed
rm -rf msfa
git clone https://codeberg.org/dcoredump/Synth_Dexed.git
mv Makefile Makefile.old
echo <<EOF >Makefile
#
# Makefile
#

CIRCLE_STDLIB_DIR = ../circle-stdlib
SYNTH_DEXED_DIR = Synth_Dexed/src

OBJS    = main.o kernel.o miniorgan.o \
          $(SYNTH_DEXED_DIR)/synth_dexed.o \

INCLUDE += -I $(SYNTH_DEXED_DIR) -I tuning-library

EXTRACLEAN = $(SYNTH_DEXED_DIR)/*.o $(SYNTH_DEXED_DIR)/*.d

include ./Rules.mk

%.o: %.cc
        @echo "  CPP   $@"
        @$(CPP) $(CPPFLAGS) -c -o $@ $<

%.d: %.cc
        @$(CPP) $(CPPFLAGS) -M -MG -MT $*.o -MT $@ -MF $@ $<
EOF
```

rsta2, 2022-02-08:
@dcoredump A replacement for `millis()` can be (`CLOCKHZ` is normally 1000000):

```
#include <circle/timer.h>

unsigned millis (void)
{
	return CTimer::Get ()->GetClockTicks () / (CLOCKHZ / 1000);
}
```

Please give more info, what the problem with the constructors is, if help is wanted there.

dcoredump, 2022-02-09:
Thanks @rsta2! I used  your `millis()` implementation and for now it seems to work.

> Please give more info, what the problem with the constructors is, if help is wanted there.

Help would be very good, because I think I'm not getting anywhere at the moment.

My problem is: I am trying to inherit for the class `AusioSynthDexed` from the two classes `Dexed` and `SOUND_CLASS`. The classes are defined in `Synth_Dexed/src/synth_dexed.h:1394`:

```
class AudioSynthDexed : public Dexed, public SOUND_CLASS
{
  public:
    AudioSynthDexed(uint8_t max_notes,uint16_t sample_rate,CInterruptSystem *pInterrupt, CI2CMaster *pI2CMaster);
    ~AudioSynthDexed(void);
    unsigned GetChunk (u32 *pBuffer, unsigned nChunkSize);
};
```

The implementation is in `Synth_Dexed/src/synth_dexed.cpp:1794`:
```

AudioSynthDexed::AudioSynthDexed(uint8_t max_notes, uint16_t sample_rate, CInterruptSystem *pInterrupt, CI2CMaster *pI2CMaster).
{
  Dexed(max_notes,int(sample_rate));
  SOUND_CLASS (pInterrupt, SAMPLE_RATE, CHUNK_SIZE
#ifdef USE_I2S
    , FALSE, pI2CMaster, DAC_I2C_ADDRESS
#endif
  );
}
```

But something is wrong with the constructor. I can't manage to call the two constructors of the parent class. I always get the following error:

```
...
  CPP Synth_Dexed/src/synth_dexed.o
Synth_Dexed/src/synth_dexed.cpp: In constructor 'AudioSynthDexed::AudioSynthDexed(uint8_t, uint16_t, CInterruptSystem*, CI2CMaster*)':
Synth_Dexed/src/synth_dexed.cpp:1794:127: error: no matching function for call to 'Dexed::Dexed()''.
 1794 | AudioSynthDexed::AudioSynthDexed(uint8_t max_notes, uint16_t sample_rate, CInterruptSystem *pInterrupt, CI2CMaster *pI2CMaster)
      | ^
Synth_Dexed/src/synth_dexed.cpp:284:1: note: candidate: 'Dexed::Dexed(uint8_t, int)'
  284 | Dexed::Dexed(uint8_t maxnotes, int rate)
      | ^~~~~
Synth_Dexed/src/synth_dexed.cpp:284:1: note: candidate expects 2 arguments, 0 provided
In file included from Synth_Dexed/src/synth_dexed.cpp:1:
Synth_Dexed/src/synth_dexed.h:1192:7: note: candidate: 'constexpr Dexed::Dexed(const Dexed&)'
 1192 | class Dexed
      | ^~~~~
Synth_Dexed/src/synth_dexed.h:1192:7: note: candidate expects 1 argument, 0 provided
Synth_Dexed/src/synth_dexed.cpp:1794:127: error: no matching function for call to 'CPWMSoundBaseDevice::CPWMSoundBaseDevice()''
 1794 | AudioSynthDexed::AudioSynthDexed(uint8_t max_notes, uint16_t sample_rate, CInterruptSystem *pInterrupt, CI2CMaster *pI2CMaster)
      | ^
In file included from Synth_Dexed/src/synth_dexed.h:56,
                 from Synth_Dexed/src/synth_dexed.cpp:1:
../circle-stdlib/libs/circle/include/circle/pwmsoundbasedevice.h:48:2: note: candidate: 'CPWMSoundBaseDevice::CPWMSoundBaseDevice(CInterruptSystem*, unsigned int, unsigned int)'
   48 | CPWMSoundBaseDevice (CInterruptSystem *pInterrupt,
      | ^~~~~~~~~~~~~~~~~~~
../circle-stdlib/libs/circle/include/circle/pwmsoundbasedevice.h:48:2: note: candidate expects 3 arguments, 0 provided
../circle-stdlib/libs/circle/include/circle/pwmsoundbasedevice.h:41:7: note: candidate: 'constexpr CPWMSoundBaseDevice::CPWMSoundBaseDevice(const CPWMSoundBaseDevice&)'
   41 | class CPWMSoundBaseDevice : public CSoundBaseDevice /// Low level access to the PWM sound device
      | ^~~~~~~~~~~~~~~~~~~
../circle-stdlib/libs/circle/include/circle/pwmsoundbasedevice.h:41:7: note: candidate expects 1 argument, 0 provided
Synth_Dexed/src/synth_dexed.cpp:1796:42: error: cannot call constructor 'Dexed::Dexed' directly [-fpermissive]
 1796 | Dexed::Dexed(max_notes,int(sample_rate));
      | ^
Synth_Dexed/src/synth_dexed.cpp:1796:42: note: for a function-style cast, remove the redundant '::Dexed'.
```

Do you have any idea what I am doing wrong?

rsta2, 2022-02-09:
@dcoredump You are welcome. Please try the following in *Synth_Dexed/src/synth_dexed.cpp*:

```
AudioSynthDexed::AudioSynthDexed(uint8_t max_notes, uint16_t sample_rate, CInterruptSystem *pInterrupt, CI2CMaster *pI2CMaster)
:  Dexed(max_notes,(int)sample_rate),
  SOUND_CLASS (pInterrupt, SAMPLE_RATE, CHUNK_SIZE
#ifdef USE_I2S
    , FALSE, pI2CMaster, DAC_I2C_ADDRESS
#endif
  )
{
}
```

The constructors will be specified after a colon after the prototype and before the function body. Multiple constructors will be delimited with a comma.

dcoredump, 2022-02-09:
@rsta2 

Perfect!

```
 CLEAN  /home/wirtz/src/MiniDexed/src
  CPP   main.o
  CPP   kernel.o
  CPP   miniorgan.o
  CPP   Synth_Dexed/src/synth_dexed.o
  LD    kernel8-rpi4.elf
  DUMP  kernel8-rpi4.lst
  COPY  kernel8-rpi4.img
  WC    kernel8-rpi4.img => 1299920
```

The next step is to initialize the object...

probonopd, 2022-02-09:
> You can include the whole thing by deleting the msfa directory and cloning SynthDexed

Thanks @dcoredump. I have updated https://github.com/probonopd/MiniDexed accordingly.

dcoredump, 2022-02-10:
Another question @rsta2:

On Teensy the audio backend works with int16_t. It seems like circle is using uint32_t.
So the question is: is a zero signal at 1<<15 and is the signal bandwith really 32 bits?

If so I had to convert the audio block like this:

uint32_signal[i]=(int16_signal[i]+0xffff)<<15;

dcoredump, 2022-02-14:
@rsta Thanks! Now I understand this strange code in minirogan :) Of course it makes sense to do it this way. I am used to a unified audio interface from the Teensy and was a bit confused.

Another simple question: Is there any macro I can put in my `Synth_Dexed` library that will tell me that the code is translated for circle? Currently I have built my own `#define USE_CIRCLE`, but of course it would be nicer to get this flag set externally.

@probonopd @rsta2 
Unfortunately I can only work sporadically, because I have other things to do, so that the fridge can be filled with beer.

rsta2, 2022-02-14:
@dcoredump Yes, such macro exists: `__circle__` (two underlines before and behind). It is defined in any file, which is translated for Circle.

No problem, just take the time you need.

dcoredump, 2022-02-15:
I've now got the whole thing working in @probonopd  fork so I think it should run (according to Murphy it won't, but whatever). My fork: `https://github.com/dcoredump/MiniDexed`. This will then need `Synth_Dexed` from [https://codeberg.org/dcoredump/Synth_Dexed](https://codeberg.org/dcoredump/Synth_Dexed).

PR to @probonopd  is also out.

Unfortunately I don't have a free Raspi for testing. That will be something towards the weekend. _If_ it works, you should hear a FM piano on some audio output triggered by MIDI-USB/Serial or keyboard. I have taken over some things from midiorgan.

rsta2, 2022-02-15:
Thank you @dcoredump! I did a quick test and with some small modifications, I can play on a USB audio class MIDI keyboard using a Raspberry Pi 4B. That's great!

You find the patches attached. Currently it does work with a MIDI keyboard only and only via the headphone jack with PWM.

[patch.zip](https://github.com/rsta2/circle/files/8073743/patch.zip)

probonopd, 2022-02-15:
Thanks a lot @dcoredump @rsta2, new build is up on
https://github.com/probonopd/MiniDexed/releases/tag/continuous

I can hear sounds when I play on my MIDI keyboard. It's not one of the well-known DX7 sounds so I am not 100% sure where it is coming from but I assume it's Dexed :+1:

dcoredump, 2022-02-18:
So, I can report that it works on a RPi4. I loaded a simple FM-Piano - the typical sound :)
Look here: https://youtu.be/9Guy3S3MUVw

rsta2, 2022-02-19:
@dcoredump Sounds good! :) How can this FM-Piano sound loaded into Synth_Dexed?

probonopd, 2022-02-19:
This rocks! What a progress. Totally awesome @dcoredump @rsta2

dcoredump, 2022-02-19:
> @dcoredump Sounds good! :) How can this FM-Piano sound loaded into Synth_Dexed?

I havn't checked this in into my repo, sorry. The way to do this is shown in
https://codeberg.org/dcoredump/Synth_Dexed/src/branch/master/examples

There is also a tool for convert sysex files into compatible C-header files inside this repo.

probonopd, 2022-02-19:
Added loading `fmpiano_sysex`. Works! :+1: 

Dexed can read `.syx` files. And Circle can read files from microSD card. So I am wondering what would be a good way to make MiniDexed work with `.syx` files.

Many USB keyboards can send Bank LSB and Bank MSB Changes, and Program Changes. If we put several `.syx` files on the microSD card, then we might use Bank LSB and Bank MSB Changes to select those `.syx` files, and Program Changes to select patches. Would that make sense?

rsta2, 2022-02-19:
I was able to test the FM-Piano too, of course a much more exciting sound than the init voice. ;)

With **circle-stdlib** you should be able to simply use `fopen()`, `fread()`, `fclose()` to access the µSD card. I don't want to meddle regarding the use cases of MiniDexed, because I'm not a musician or synth expert. But because the MIDI interface code is from the sample/29-miniorgan, I suppose, SysEx transfers will not work yet.

probonopd, 2022-02-19:
Very weird: I mentioned earlier that USB is flaky on my device.

Now I found out that if no display is attached to the HDMI 1 port, then I always get `xhcidev: Cannot set device address (19)`. (I can see this message when I attach a display to HDMI port 2.)

Tried with the original Raspberry Pi 4 Power supply and with another USB-C one.

Does it work for you when you have no display attached?

dcoredump, 2022-02-19:
> Added loading `fmpiano_sysex`. Works! :+1: 
> 
> Dexed can read `.syx` files. And Circle can read files from microSD card. So I am wondering what would be a good way to make MiniDexed work with `.syx` files.
> 
> Many USB keyboards can send Bank LSB and Bank MSB Changes, and Program Changes. If we put several `.syx` files on the microSD card, then we might use Bank LSB and Bank MSB Changes to select those `.syx` files, and Program Changes to select patches. Would that make sense?
> 
> In addition, we may want to listen to and use sysex messages sent over MIDI. Or is this already the default behavior? (Couldn't test due to the lack of a USB MIDI adapter that I could use for sending sysex from Dexed on the PC to MiniDexed.)
> 
> @dcoredump I would like to print out MIDI messages for debugging around [here](https://github.com/probonopd/MiniDexed/blob/4fe9038a836a8d0a0d79c52f2a1865f258c91502/src/kernel.cpp#L60) and possibly react to those MIDI messages in an if/switch statement to load different default sysex - is this possible?

Sysex will work, but let me put the MIDI code (with sysex support) from MicroDexed into Synth_Dexed, so all projects using the library will get full sysex support.

MicroDexed can currently handle all: beside the normal MIDI implementation also voice upload and voice parameter changes.

rsta2, 2022-02-19:
@probonopd I copied the files from your MiniDexed Releases page to an empty µSD card and tried it with a display connected to the second HDMI port. No problems.

What Raspberry Pi 4 model do you have? I have a very early RPi 4B 2GB version. Do you use an USB hub or any other USB devices beside the USB MIDI keyboard?

Currently there must be a display attached to one of the HDMI ports in any case on the Raspberry Pi 4. This is different on Raspberry Pi 3, which works without a display. If you want to run without a display, you have to add the following line to the file *circle-stdlib/libs/circle/Config.mk* after running *configure*:

```
DEFINE += -DSCREEN_HEADLESS
```

There will be no display output then, even when a display is attached.

probonopd, 2022-02-19:
I have a very early RPi 4B 1GB but updated the flash ROM. The only combination that works for me is when I plug in a display in HDMI port 1, the Original Raspberry Pi keyboard into a USB 2 port, and the USB MIDI keyboard into the Original Raspberry Pi keyboard. At this point I cannot entirely exclude the possibility of some of my hardware being particular/faulty.

[5 more replies]

closed 2022-02-20 by probonopd