rsta2/circle #275: SPI with software ChipSelect

sebastienNEC, 2022-01-31:
Hello, sometimes people need to control more than 2 SPI chips, while the SPI hardware can only control 2 ChipSelect lines (CE0/CE1). Would you be interested in a patch to allow software ChipSelect in CSPIMaster?
If so, I am not sure of the semantics. For now, I just added a boolean m_bSoftChipSelect to the constructor and allowed nChipSelect to refer to a gpio number in the Read/Write() methods.

rsta2, 2022-02-01:
Isn't it possible, to do this soft-chip-select handling outside of the `CSPIMaster` class? The user can easily use the `CGPIOPin` class to define the number of soft-CS-pins, he needs, and write LOW to it before calling the transfer function of the SPI master. After the transfer he writes HIGH to the soft-CS-pin. I think, this is more flexible, because we do not know, how many soft-CS-pins are needed in an application.

sebastienNEC, 2022-02-01:
The problem is that the current driver forces nChipSelect to be 0 or 1 in the Read/Write methods. So if the user had SPI chips on e.g. CE0 and another gpio, then both chip select lines would be LOW and both chips would engage their MOSI/MISO lines...

Manually toggling the CS GPIO is possible outside of CSPIMaster indeed, but then we'd still need a way to tell this class to not touch the CE0/CE1 pins e.g. m_bSoftChipSelect in the constructor?

rsta2, 2022-02-01:
Yes, I understand. I would suggest to extend the valid range for `nChipSelect` to 0..2, which only requires an assertion to be modified. There can be a constant `CSPIMaster::ChipSelectNone`, which has the value 2. If one specifies this value to one of the transfer methods, the lines CE0 and CE1 should remain inactive and CE2 is not exposed for the SPI 0 master, but can be selected in the SPI_CS register.

sebastienNEC, 2022-02-02:
Ok this would work indeed, with a minor inconvenience: if one uses the CE0/CE1 pins for something else (e.g. a gpio input interrupt in my case), one has to change the gpio's mode with SetMode() after the SPI constructor/destructor. But this is definitely manageable.
I think the value of nChipSelect would be 3, not 2. At least this is how they do it in the linux driver. The changes to the class would then be in 2 lines. The assert that you mentioned, but also CS_CS:
`#define CS_CS		(3 << 0)`
Thank you !

rsta2, 2022-02-03:
OK, I will implement this soon. Thank you for the corrections!

rsta2, 2022-03-11:
This modification has been released with Circle 44.4.

closed 2022-03-11 by rsta2