rsta2/circle #276: Second Memory Interface

sebastienNEC, 2022-02-04:
I have to drive a strip of Neopixel LEDs but my SPI/PWM pins are taken for other functions, so I thought of using the SMI interface as described here: https://iosoft.blog/2020/09/29/raspberry-pi-multi-channel-ws2812/

I've therefore tried to write an SMI driver for circle. As always, I am not sure of the semantics, so any comments welcome :)
[smi.zip](https://github.com/rsta2/circle/files/8004185/smi.zip)

rsta2, 2022-02-05:
I need to read about SMI, before I can express an opinion. This will take some time.

rsta2, 2022-02-20:
@sebastienNEC Do you have any problem with the requested modifications?

sebastienNEC, 2022-02-21:
Hello Rene, sorry for not answering soon enough, I had a rough 2 weeks :)
I don't have any problems with the mods and I'll implement them soon ! Any particular deadline from your side?
I have tested on a raspberry pi zero W

rsta2, 2022-02-21:
Ah OK, no problem, take the time you need. There is currently no deadline.

rsta2, 2022-02-23:
I added your code to the *smi-support* feature branch. Great work! I will "tune" it a little bit and add the sample. I will probably need you to test the sample again. Thank you very much!

rsta2, 2022-02-24:
I added the sample to *addon/WS28XX/multichan/*. It supports three NeoPixels at GPIO8-10. I was able to test it successfully on a Raspberry Pi 4B. It would be great, if you could test it with your hardware.

I found a problem with the handling of the data cache. You had probably noticed this already, because there was a "todo" in the source code. The problem is solved, when `CDMAChannel::SetupIOWrite()` is called prior to each DMA transfer, not only once. There is not much overhead, when doing this.

I also removed the pointer to `CInterruptSystem` in the constructors. `CDMAChannel` does not need this pointer, if the completion routine is not used, like it is the case in your code.

sebastienNEC, 2022-02-28:
Aha interesting that SetupIOWrite() fixes that bug. My guess is that it is related to the cache. Maybe CleanAndInvalidateDataCacheRange() would be enough ?

rsta2, 2022-02-28:
Yes, it is cache related. Manipulating the data cache directly would probably work too. But this is not the intended use of the CDMAChannel class. By intention each DMA transfer is initiated with calls to `Setup*()`, maybe `SetCompletionRoutine()` and `Start()`. Then `Wait()` should be called, or `GetStatus()` from the completion routine, if one has been set before. This order is used throughout Circle. Also I think it's better to hide the cache maintenance routines from the user as far, it is possible.

rsta2, 2022-02-28:
I added a Doxygen doc for `CDMAChannel` on the *develop* branch, which explains the intended use of it. I think this doc was not sufficient before.

rsta2, 2022-03-06:
The multi-channel NeoPixel and SMI support has been merged to the *develop* branch to be part of the next release. The *smi-support* branch has been removed. In the meantime I was able to test the NeoPixel sample successfully on several RPi models, including the Zero W. Thanks again for this contribution!

sebastienNEC, 2022-03-07:
Great, thank you Rene !

rsta2, 2022-03-11:
The SMI and multi-channel NeoPixel support is in Circle 44.4 now.

closed 2022-03-11 by rsta2