rsta2/circle #277: Feature request: add support for lua

stephaneweg, 2022-02-05:
Hi,
i dont know how to send a feature request here, that's why.
I think it would be great to have the LUA VM ported as addon for circle

there is already a project in github that have successfully ported lua to bare metall raspberry pi:
https://github.com/leiradel/barebones-rpi/tree/master/barebones08/lua

i do not really know how i could port it in the addon . 
tank you

olfp, 2022-02-06:
You may want to have a look at 
https://github.com/olfp/lupos

This is a effort by me to run lud on top of circle. It provides a basic REPL an editor and file system access. Tested oh Pi 3b+ and Pi 400.

It‘s now inactive due to lack of time :-(

rsta2, 2022-02-07:
@stephaneweg I think, this is an interesting feature, but out of the scope of Circle itself. Unfortunately I have not the time and the Lua knowledge for such a separate project. Perhaps the project of @olfp can help you.

rsta2, 2022-02-20:
As I told, I think, this is out of scope of Circle. That's why I close this issue.

closed 2022-02-20 by rsta2