rsta2/circle #278: RPI Wired Keyboard 2 not working

stephaneweg, 2022-02-18:
(maybe related to issue 273)

i am trying to test the example 08 (usb keyboard)
i have a raspberry pi official keyboard

when i plug the keyboard, it seems to be correctly detected

usbdev1-2: Device ven5e3-610, dev9-0-2 found
usbdev1-2: product : usb2.0 hub
usbdev1-2: interface int9-0-1 found
usbdev1-2: function is not supported
usbdev1-2: interface int9-0-2 found
usbdev1-2: using device/interface int9-0-2
usbdev1-2-1: Device ven4d9-6 found
usbdev1-2-1: Product:  RPI Wired Keyboard 2
usbdev1-2-1: interface int3-1-1 found
usbdev1-2-1: usign device/interface int3-1-1
usbdev1-2-1: interface int3-0-0 found
usbdev1-2-1: usign device/interface int3-0-0
usbhub: Port 1: Device configured
usbhub: Port 2: Device configured

when i type something it does'nt seem to detect anything

there is my cmdline.txt:
dwc_otg.lpm_enable=0 root=PARTUUID=7A0C8BB0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes usbhid.mousepoll=0 rootwait

(i have removed usbignore=int3-1-1, because the keyboard seems to use it, but when i remove it, my mouse does not work)

stephaneweg, 2022-02-18:
I managed to implement a patch

step 1 : add a method "FindString" in the CString class (strstr is fine)
step 2: in the UsbDeviceFactory class: when it try to returns the keyboard for the interface 3-1-1, replace the existing content by 
else if (pName->Compare ("int3-1-1") == 0)
	{
		CString productName = pParent->GetDevice()->GetProduct();
		if (productName.FindString("Keyboard"))
		{
			pResult = new CUSBKeyboardDevice (pParent);
		}
		
		else
		{
			CLogger::Get ()->Write ("ufactory", LogWarning,
					"Ignoring interface %s (Not a keyboard)", pUSBIgnore);
		}
	}

step3: remove the "usbignore=int3-1-1" from the cmdline.txt file
after that, my mouse and my keyboard worked together correctly

rsta2, 2022-02-18:
Yes, with the mouse workaround, suggested in issue #273, it's not possible to use an USB keyboard parallel to the mouse. Thanks for suggesting a patch, which works so that you can use your keyboard and mouse together. Unfortunately some USB keyboards do not have the string "Keyboard" in the product string or do not have a product string at all. Because there are more of such workarounds in the meantime, it is necessary to implement a more general solution for this problem. I have to think about this.

rsta2, 2022-02-19:
I added a solution, which is quiet similar to yours on the *develop* branch. It uses the vendor and product ID to detect your mouse and to prevent loading a driver for the USB keyboard interface then. Is it possible to test this with your mouse and keyboard? Thanks for giving the hint!

stephaneweg, 2022-03-01:
your workaround works for me, when i plug my mouse, it does not try to return a keyboard device

rsta2, 2022-03-01:
Good to know. Thanks for testing!

rsta2, 2022-03-11:
This workaround is in Circle 44.4.

closed 2022-03-11 by rsta2