rsta2/circle #279: Booting with u-Boot

doraemoncito, 2022-02-26:
Hi, I'm trying to run the example code from the samples directory in circle over TFTP using u-boot on a Pi4 and I'm finding that many of them hang / freeze.  It I get u-boot to download itself over TFTP and run, it just works, e.g. u-boot happily boots u-boot so I'm working if there is something I'm missing during the compilation of the circle samples that might be causing this.

This is the console log output from u-boot when starting the gamepad sample:

```text
[STAGE 2] ===== Downloading ZX Raspberry boot image from TFTP server =====
Using ethernet@7d580000 device
TFTP from server 192.168.1.10; our IP address is 192.168.2.3; sending through gateway 192.168.2.2
Filename 'kernel7l.uimg'.
Load address: 0x8000
Loading: #########################################  0 Bytes
         821.3 KiB/s
done
Bytes transferred = 206272 (325c0 hex)
[STAGE 2] ===== Launching ZX Raspberry kernel image =====
## Booting kernel from Legacy Image at 00008000 ...
   Image Name:   Circle sample
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    206208 Bytes = 201.4 KiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 2eff2f00
   Booting using the fdt blob at 0x2eff2f00
   Loading Kernel Image
   Using Device Tree in place at 2eff2f00, end 2f002f0d

Starting kernel ...

logger: Circle 44.3 started on Raspberry Pi 4 Model B
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.73 pcie: Link up, 5.0 Gbps x1
00:00:02.79 usbdev1: Device ven2109-3431, dev9-0-1 found
00:00:02.79 usbdev1: Product: USB2.0 Hub
00:00:02.80 usbdev1: Interface int9-0-0 found
00:00:02.80 usbdev1: Using device/interface int9-0-0
00:00:02.80 except: stack[7] is 0xFE38
00:00:02.80 except: stack[21] is 0x2B13C
00:00:02.80 except: stack[30] is 0x1F000
00:00:02.80 except: stack[31] is 0xDCA8
00:00:02.80 except: stack[37] is 0xD1F8
00:00:02.80 except: stack[43] is 0xB04C
00:00:02.80 except: stack[47] is 0x29700
00:00:02.80 except: stack[48] is 0x30058
00:00:02.80 except: stack[55] is 0x8000
00:00:02.80 except: stack[61] is 0x808C
00:00:02.80 except: Data abort (PC 0x9898, FSR 0x221, FAR 0x81F003, SP 0x225918, LR 0x987C, PSR 0x2000011F)
```

Any suggestions?

doraemoncito, 2022-02-26:
Here is the complete boot log: [u-boot-log.txt](https://github.com/rsta2/circle/files/8146790/u-boot-log.txt)

rsta2, 2022-02-26:
There is an update on the *develop* branch, which should fix the problem. Can you please try it? Thanks for reporting the problem.

doraemoncito, 2022-02-27:
Excellent work.  That's fixed it!

rsta2, 2022-02-27:
Great! Thanks for testing!

rsta2, 2022-03-11:
This issue has been fixed in Circle 44.4.

closed 2022-03-11 by rsta2

doraemoncito, 2022-03-12:
Awesome, thank you.