rsta2/circle #280: Rotary encoder support?

probonopd, 2022-02-27:
Is there rotary encoder support in Circle?

[image]

Together with as small display this could be used to navigate simple menu structures.

rsta2, 2022-02-27:
Not yet. Where can one favorable and is there a datasheet somewhere?

probonopd, 2022-02-27:
The [KY-040-2](https://www.berrybase.de/bauelemente/passive-bauelemente/potentiometer/drehimpulsgeber/drehregler/rotary-encoder-mit-breakoutboard-ohne-gewinde-und-mutter) is a common one.

Here is one explanation but there are plenty of examples for rotary encoders found on the web:
https://www.epitran.it/ebayDrive/datasheet/25.pdf

Maybe this code could be helpful?
https://github.com/F6HQZ/rotary-encoder-lib-for-raspberry-pi

probonopd, 2022-02-27:
One thing to be aware of:

> I spent several days fighting with the noisy encoder and trying so many different ways of debouncing the signal that I lost count. Hopefully this will help keep someone else from pulling their hair out in frustration.

https://www.reddit.com/r/raspberrypipico/comments/pacarb/sharing_some_c_code_to_read_a_rotary_encoder/

rsta2, 2022-02-27:
Should be possible to support it. I will order one.

rsta2, 2022-02-28:
My KY-040 should arrive until Wednesday. I already prepared a driver, which waits for testing. Implementing a user interface with the LCD and the rotary encoder will probably be a lot of work.

probonopd, 2022-03-01:
...unless we can leverage existing work from, e.g., [MicroDexed](https://codeberg.org/dcoredump/MicroDexed)?

rsta2, 2022-03-01:
Ah yes, this can be a great help.

probonopd, 2022-03-02:
Note to self: 

```
./configure
./makeall
cd ./addon/sensor
make
cd sample/ky040
make
```

rsta2, 2022-03-02:
You are welcome. I tried it on the RPi 2B with this [KY-040](https://www.azdelivery.de/en/products/drehimpulsgeber-modul?_pos=1) without problems. Have you cleaned the project using `makeall clean` and `make clean`, if you used a different RPi model before? Also `configure` does not overwrite an existing configuration. Maybe you have to use `configure -f -r 2`?

I will see, what I can do for the additional button functions.

rsta2, 2022-03-03:
I have added the suggested events (if I understood them right).

I also added a polling mode, for the case we must run the MiniDexed UI on a secondary core, because there is already much interrupt load on core 0, and the KY-040 driver requires a small interrupt latency.

Please let me know, if you have more suggestions.

rsta2, 2022-03-04:
Tried to use the KY-040 together with MiniDexed. It works! I can switch between program change and bank select with one click and change the voices or banks by turning the knob. Pressing the switch for 3 seconds reboots the system.

I did a mistake first and used the same GPIO pin 19 for the LCD display and the KY-040. I was guessing, the interrupt latency is too high, so that the rotary encoder does not work with interrupts. But luckily this is not the case. Even on a RPi 1A+ it does work.

@probonopd This is a quick hack at them moment. I don't know, if I should develop a clean solution and send a PR? If so, which 3 GPIO pins should we use for the encoder and switch by default in the configuration?

probonopd, 2022-03-04:
Ideally we find a combination of pins that allows Serial MIDI, display, rotary encoder, and DAC all to work.

Unfortunately I have no access to my hardware for the next few days. Will try your suggestions above asap. Thanks!

probonopd, 2022-03-08:
@rsta2 I was now able to test the implementation and it is working for me. Just one nitpick: I have to turn the knob ~2-3 physical positions until it results in a change. I am not sure whether this varies between different models of rotary encoders or even between different devices of the same model. Maybe we should make it configurable? (3D printer firmware Marlin has a setting `#define ENCODER_PULSES_PER_STEP 1` for this.)

rsta2, 2022-03-08:
@probonopd Good to hear, that it is working for you. :) Thanks for info. Unfortunately currently I don't know, how to make the rotary encoder more sensitive. Perhaps the polling driver gives better results here, but we would need multi-core for this in MiniDexed.

rsta2, 2022-03-11:
The KY-040 driver has been released with Circle 44.4.

probonopd, 2022-03-11:
Thank you very much @rsta2. This is highly appreciated :+1:

rsta2, 2022-03-13:
I think this can be closed now.

closed 2022-03-13 by rsta2