rsta2/circle #282: No mmap, trying to write a JIT - how to mark memory allocation as executable?

zolbatar, 2022-03-18:
I'm trying to embed GNU Lightning to do some JIT stuff, but had to remove the mmap stuff as there isn't a library for it in circle.

When I call the generated machine code, it crashes out, presumably because heap memory isn't executable.

I've used GDB, and it successfully jumps to the first instruction but then if that is attempted to execute it drop back to presumably an exception handler. If QEMU is used with -semihosting it closes QEMU. 

I've tried altering the armv6mmu.h, armv7lpae.h and armv8mmu.h respectively and not got a change.

Any ideas?

rsta2, 2022-03-18:
Yes, memory regions above `_etext` are normally marked as "execute never" (XN). This is implemented in *lib/pagetable.cpp* and *lib/translationtable64.cpp*. Please search for any symbol, which contains an "XN" in these files, and if a number 1 is assigned, replace it with 0. `ARMV6MMUL1SECTION_NORMAL_XN` has to be replaced with `ARMV6MMUL1SECTION_NORMAL`.

zolbatar, 2022-03-19:
Perfect, works a treat! Thank you.