rsta2/circle #283: UART - problems - no life

duaneellissd, 2022-03-21:
What Works:

09-softserial - HDMI screen works, I can print things on the screen :-)

But nothing comes out the serial port.

TARGET:  "raspberry pi 3 model B+" - is what it shows on the PCB
Exactly this: (pictures match). https://static.raspberrypi.org/files/product-briefs/Raspberry-Pi-Model-Bplus-Product-Brief.pdf

SERIAL Cable - (Adafruit 954) connected exactly like this:
https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/connect-the-lead

My hunch: There is something goofy with the clocks, etc

To verify the UART is working correctly -
AT FIRST - I could not get the SERIAL PROMPT (ie: the serial login prompt)
Eventually, I found a rpi forum that said add these items to the config.txt file:

arm_freq=700
arm_freq_min=700
core_freq=250
force_turbo=1
enable_uart=1

Once I did that, the serial prompt worked and I got a SERIAL LINUX LOGIN prompt

Ok - that in my view proves the serial cable is correct.

HOWEVER Using any "circle" thing I cannot get the UART to work.
All that I do is replace the "kernel8.img" with the image from circle

I added a number of DEBUG print statements on the HDMI monitor, it reports on the HDMI monitor:
Circle 44.4.1, and I am using AARCH64 - not 32bit
Speed Factor 1.75
uart clock = 48,000,000
Emmc clock = 200,000,000
Emmc2 clock = 100,000,000
core clock = 250,000,000
pixel clock = 0,
machine name: Raspberry Pi 3 Model B+
Model: 0x0000000c. (decimal 12)
SocType: 0x00000002 (decimal 2)
Ram Size: 0x00000400, I believe this is MegBytes, so this would be 1GIG
SOC Name: BCM2837
RawRev: 0x00a020d3

I'm sort of stuck and can't easily make further progress,
do you have a suggestion what I can try?

rsta2, 2022-03-21:
By default most Circle samples do not send data to the UART. You can enable sending the log output to the UART with this setting in the file *cmdline.txt* on the SD card:

```
logdev=ttyS1
```

For example *sample/04-timer* will with this send the log output at 115200 Bps 8-N-1 to the UART.

As a test, please build and run *sample/32-i2cshell* without a connected USB keyboard. It will then offer a simple shell on via the UART (same parameters as above). I tried this sample on the same RPi model and it works. You do not need special settings in *config.txt* for this. Enabling the 64-bit mode and setting the kernel image name (as in *boot/config64.txt*) are enough.

It's always a good idea to use the firmware, which can be downloaded in *boot/* using `make`, because Circle will be tested with this firmware.

duaneellissd, 2022-03-22:
yea, that works - but does not answer some fundamental items I have.
A) it seems it should have worked, ie: this boot code is doing something different.
I would think the app loaded should configure the clocks correctly, or get the correct values from the clocks.
And should be able to configure the UART so that it works
Sigh.. it's not going to be .... You move on...

=======

And for those reading this or finding it via GOOGLE...

My original problem was:  The UART did not work with Circle.
I found some settings that made it partly work - but only with default linux rbi-stuff
(I wanted a bare metal solution instead)
When I tried it did not work - the above exchange helped me figure it out.

The mystery item seems to be the content of the SD CARD what worked for me is this:

step 1 - I have an 8gig card, formatted in windows format 
Step 2 - the files (and sources are listed below)

* kernel8.img -> Result of 'make' in the 32-i2cshell directory
* config.txt -> renamed from boot/config64.txt
* bcm2711-rpi-cm4.dtb <- Result of "make" in the boot directory
* bcm2711-rpi-4-b.dtb <- Result of "make" in the boot directory
* bcm2711-rpi-400.dtb <- Result of "make" in the boot directory
* bootcode.bin <- result of "make" in the boot directory
* start.elf <- result of "make" in the boot directory
* start_cd.elf  <- result of "make" in the boot directory
* start4cd.elf <- result of "make" in the boot directory
* fixup.dat  <- result of "make" in the boot directory
* fixup_cd.dat  <- result of "make" in the boot directory
* fixup4.dat  <- result of "make" in the boot directory
* fixup4cd.dat  <- result of "make" in the boot directory

* In my case, I have one more directory: ".Spotlight-v100" - I work on a Mac using Ubuntu via Parallels the Mac creates these directories..

* I am using "cool term" on the Mac (I don't have putty or tera term), works nicely but I also like Tera Term on windows
* Oh, and Serial settings are 115,200 NoParity, 8 bits, 1 stop bit

END.

closed 2022-03-22 by duaneellissd

rsta2, 2022-03-22:
There are incompatible modifications to the firmware from time to time. It's not possible to react on each firmware change immediately. That's why there is the recommended firmware version, downloadable in *boot/*, which is tested with Circle.