rsta2/circle #285: Problem with emmc detecting SDCard on CM3+

randyrossi, 2022-04-06:
Hi.  I have someone using my BMC64 release on a Compute Module 3+ but the emmc module is unable to detect the sdcard.  Here is the logging with extra debug for emmc turned on.  I was wondering if you had any ideas why this is happening?  (I am using an older version of circle : hash fe24b6bebd1532f2a0ee981af12eaf50cc9e97fb   release-40, but CM3+ appears to be supported in that release).  Thanks.

logger: Circle 40 started on Compute Module 3+
 
mcore: CPU core 1 started
 
mcore: CPU core 2 started
 
Core waiting for launch
 
mcore: CPU core 3 started
 
printk: vchiq: vchiq_init_state: slot_zero = 0x02280000, is_master = 0
 
00:00:00.02 emmc: Vendor 99, SD version 2, slot status 0
 
00:00:00.02 emmc: Resetting controller
 
00:00:00.03 emmc: control0: 00000000, control1: 00000000, control2: 00000000
 
00:00:00.03 emmc: checking for an inserted card
 
00:00:00.04 emmc: status: 1fff0000
 
00:00:00.04 emmc: Base clock rate is 250000000 Hz
 
00:00:00.04 emmc: setting clock rate
 
00:00:00.05 emmc: base_clock: 250000000, target_rate: 400000, divisor: 00000200, actual_clock: 244140, ret: 00000080
 
00:00:00.06 emmc: control0: 00000000, control1: 000b0083
 
00:00:00.06 emmc: enabling SD clock
 
00:00:00.07 emmc: card insertion detected
 
00:00:00.08 emmc: Issuing command CMD0
 
00:00:00.08 emmc: command completed successfully
 
00:00:00.08 emmc: Note a timeout error on the following command (CMD8) is normal and expected if the SD card version is less than 2.0
 
00:00:00.09 emmc: Issuing command CMD8
 
00:00:00.10 emmc: Error occured whilst waiting for command complete interrupt
 
00:00:00.11 emmc: Error issuing CMD8 (intr 00018000)
 
00:00:00.11 emmc: CMD_TIMEOUT
 
00:00:00.11 emmc: Note that a timeout error on the following command (CMD5) is normal and expected if the card is not a SDIO card.
 
00:00:00.12 emmc: Issuing command CMD5
 
00:00:00.13 emmc: Error occured whilst waiting for command complete interrupt
 
00:00:00.13 emmc: Error issuing CMD5 (intr 00018000)
 
00:00:00.14 emmc: CMD_TIMEOUT
 
00:00:00.14 emmc: sending inquiry ACMD41
 
00:00:00.15 emmc: Issuing command ACMD41
 
00:00:00.15 emmc: Error occured whilst waiting for command complete interrupt
 
00:00:00.16 emmc: Error issuing CMD55 (intr 00018000)
 
00:00:00.16 emmc: CMD_TIMEOUT
 
00:00:00.16 emmc: Inquiry ACMD41 failed
 
Core 3 idle

rsta2, 2022-04-06:
The embedded MMC is unfortunately not supported on the CM1 and CM3(+). I never had these devices here and that's why their support is limited. Therefore they are listed in parentheses in the [list of RPi models](https://github.com/rsta2/circle/blob/master/README.md?plain=1#L164) in the main README.md.

randyrossi, 2022-04-06:
Thanks.  Is there a way to use the SD1 interface instead?  That is, they can flash the emmc with the kernel for boot but then have circle mount the sdcard for everything else the code needs to access. 

The user is using a CM3+ as their prototype but their end design will use a CM3+ Lite with SDcard slot populated instead of the emmc flash.  So this would just be a way to get my program to run on the CM3+ for dev purposes.  (As far as I know, there is no way to boot from the sdcard slot on a CM3+)

rsta2, 2022-04-06:
Unfortunately I don't know. You can only test, if it works, if you have a CM3+ (Lite), for example with the sample program in *addon/SDCard/sample/*. Newer Circle versions (since Step42) have two variants of the SD card driver. By default the system option `USE_SDHOST` is enabled there, which uses the SDHOST interface, instead of the EMMC interface. If you explictly define `NO_SDHOST` there, the EMMC interface is used. But before Step42, the EMMC interface is the only option and is enabled by default.

randyrossi, 2022-04-06:
Thanks.  I will recommend they short EMMC_DISABLE_N to GND to force usb boot.  Since they don't need the embedded mmc anyway.

closed 2022-04-06 by randyrossi