rsta2/circle #287: 38-bootloader: Request Entity Too Large

probonopd, 2022-04-22:
https://github.com/rsta2/circle/tree/master/sample/38-bootloader is giving me `Request Entity Too Large` when trying to load MiniDexed `kernel8.img` which is 2514992 B.

probonopd, 2022-04-22:
Thanks. This works for me. Maybe it would be worthwhile to make it the default?

One side note: I was puzzled why uploading `kernel8.img` to my RPi 3 wouldn't woork. `kernel7.img` worked though.

closed 2022-04-22 by probonopd

rsta2, 2022-04-23:
2 MB is the default kernel size for all Circle kernels. By far most kernel images are smaller and increasing the `KERNEL_MAX_SIZE` parameter would waste memory for these cases.

Have you configured Circle with `AARCH=64`, when building the bootloader? The bootloader must be built for the same architecture as the kernel image, which will be loaded.

The firmware config is only observed, when loading the bootloader itself. It is the same then for the loaded kernel image. Config files of the application should be accessible as normal.