rsta2/circle #288: Small  SD cards do not work correctly

fp64lib, 2022-04-24:
This issue originally was raised for the MiniDexed. After booting the appropriate kernel image, access to the FAT32 file system does not work correctly for a small Micro SD card, i.e. in my case 128MB. More details can be found here https://github.com/probonopd/MiniDexed/issues/160 , also what was already checked. Filesystem is definitely FAT32, card was freshly formatted.

rsta2, 2022-04-25:
How many clusters does this volume has? If less than 65525, it should be formatted as FAT16. In MiniDexed the [FatFs](http://elm-chan.org/fsw/ff/00index_e.html) file system module is used for file access and I don't think, there is a bug in it regarding this, because it is widely used.

probonopd, 2022-04-25:
Learned something. Will add this to the documentation.

fp64lib, 2022-04-25:
> How many clusters does this volume has? If less than 65525, it should be formatted as FAT16. 
It's a 128MB Sd card with 240.112 clusters.

>I don't think, there is a bug in it regarding this, because it is widely used.
"I don't think there are bugs in Microsoft Windows because it is widely used." Maybe not the best line of defense, if someone has clearly documented that MiniDexed (and also MT-32) does not work on a small SD card.

rsta2, 2022-04-25:
I tried a similar sized µSD card and a very old 1.00 or 1.01 SD card with Circle. Both worked well. I cannot exclude, that there are (µ)SD cards, which do not work with Circle. I can only recommend to use an other card.

probonopd, 2022-04-26:
With the smallest ones one can buy today well in the GB range, I don't think it's worthwhile to spend a lot of effort on this. 

@fp64lib if you really care about this, maybe it'd be worthwhile to ask over at http://elm-chan.org/fsw/ff/bd/ and maybe test the same SD card with another fatfs based project?

rsta2, 2022-04-26:
@probonopd No, this may require a license fee. That's why this is disabled in Circle. See: http://elm-chan.org/fsw/ff/doc/appnote.html#exfat (second paragraph).

richard-broadhurst, 2023-06-11:
I had issues with a few of the 100+ 128MB cards I have used where the SPI return code wasn't putting the usual 111 or 000 in the unused bits which mean that it thought that a write had failed when it had actually succeeded meaning that it only did a partial update. The fix was to ignore the unused bits!
I'm not saying that this is your problem and it would affect all file systems, just mentioning in case it is helpful.

rsta2, 2023-06-11:
Thanks. In which controller register are these bits? May masking out these bits affect operation with other SD cards?

richard-broadhurst, 2023-06-11:
I think it is the write response byte, but a better explanation from the guy who fixed it is here: https://stardot.org.uk/forums/viewtopic.php?p=247106#p247106

rsta2, 2023-06-11:
I guess, we are talking about different issues here, because the initial problem was (most likely) related to the FAT file system driver. Unfortunately I do not have direct access to the data token, sent by the SD card, which were mentioned in the referenced article. I can only program the EMMC or SDHOST controller of the Raspberry Pi. The SD card is not accessed via SPI in the Circle drivers. I'm open for fixes, but generally it is difficult, because I do not have these SD cards here, which do not work. I have a  16 MB SD card from Canon is working well with Circle.

ArnoldUK, 2023-06-11:
I had similar issues with random errors and narrowed it down to the FatFs driver. After switching over to Circle CFATFileSystem everything worked perfectly. FatFs is much more complicated for me to even start bug finding but the major problems were random lockups accessing the card and initializing the card.

ArnoldUK, 2023-06-13:
I'm not sure as I only work with FAT16 partitions of sizes less than 512MB. I dont use FAT32 in my projects so the problem I had could well have been a FAT16 only problem. I have sine updated my circle build with the newer FatFs driver and so far I have not any issues with the FATFS drivers when using an SD card. I do rememebr that the older FatFs driver read clusters in 512 byte chunks and if I did not format the card with 512 byte clusters then a would get random lockups reading/writing to the card.

rsta2, 2023-06-13:
Thanks for info! So in case of problems my suggestion would be to use SD cards with partitions >= 1 GB size. Of course it would be good to find the reason of an issue, if there is one, but unfortunately I wasn't able to reproduce any problem in this field with the available error reports.