rsta2/circle #289: QEMU setup instructions seems to be outdated.

DFL1973, 2022-04-26:
I'm trying to install using Ubuntu, but fails when I try to "/configure" with a lot of errors. (error to create symbolic links, meson error and others)

rsta2, 2022-04-26:
Which QEMU did you tried? My own QEMU repo is outdated, that's true. The current version is maintained [here](https://github.com/smuehlst/qemu).

zolbatar, 2022-05-05:
I had to patch meson because I'm on a Mac and issues with case insensitive filenames and "version".

I still can't for the life of me get the USB keyboard and mouse to work though even with the patched QEMU? Any idea?

zolbatar, 2022-05-05:
I tried that and still can't get it to detect a keyboard. I wonder if the fact I am running this on macOS affects things in any way?

rsta2, 2022-05-06:
I tried the following again successfully on a PC/Linux host:

```
cd circle-stdlib
configure -r 3 -p aarch64-none-elf-
make -j
cd samples/04-std
make -j
qemu-system-aarch64 -M raspi3b -kernel ./kernel8.img \
	-drive file=/var/mtools/sdcard.img,if=sd,format=raw \
	-device usb-kbd
```

The keyboard works. Unfortunately I cannot talk about using it on macOS, because I do not have it.

zolbatar, 2022-05-06:
Thank you for trying, I might try getting it to work on Linux.

rsta2, 2022-12-04:
The `--qemu` option of [circle-stdlib](https://github.com/smuehlst/circle-stdlib) `configure` does not disable USB any more for AArch64 with Release v15.14. So this issue should be solved and can be closed.

closed 2022-12-04 by rsta2