rsta2/circle #293: RPi Pico CircuitPython USB MIDI Device not recognised by Circle on a Pi 4

diyelectromusic, 2022-05-05:
A simple DIY Raspberry Pi Pico CircuitPython USB MIDI device (apparently supporting the audio/MIDIStreaming class) does not seem to be recognised by Circle running in 32-bit mode on a Raspberry Pi 4.  For context, see this discussion from MiniDexed: https://github.com/probonopd/MiniDexed/discussions/197

I have reproduced this issue with the miniorgan sample from within circle.  I have built the following configurations:
Prefix = arm-none-eabi-
AARCH = 32
RASPPI = 1, 2, and 4

With CIRCLEVER = 440401 I have a kernel.img, kernel7.img and kernel7l.img (all built with a makeall clean followed makeall and then make clean and make in the 29-miniorgan sample directory) and have tested the same memory card (containing all three kernels) with a Raspberry Pi V1 Model B revision 2, a Raspberry Pi V3 and a Raspberry Pi V4 all with I2S PCM5102 as an output device (the Pi V1 is using the P5 header, the Pi V3 and V4 use my ClumsyMIDI HAT).

My XKey USB MIDI keyboard works on all three targets.  My DIY Pi Pico CircuitPython USB MIDI keyboard works on the V1 and V3 Pis but not on the V4.  A screenshot of the HDMI boot output is provided below. In this you can see the miniorgan sample starting, the XKey being plugged in and used successfully, then removed and the Pi Pico DIY keyboard plugged in and getting the error: "xhciep: Transfer Timed Out" when attempting to configure the int1-3-0 audio MIDIStreaming endpoint for the Pico.

I haven't been able to test the 64-bit architecture as I don't have a valid 64-bit build environment for Circle at the moment.

I haven't been able to isolate an older version of Circle to prove this conclusively, but I also have a version of MT32-Pi where the Pico USB MIDI device works fine on both a Pi V3 and Pi V4.  I believe this is MT32-Pi v0.10.3.  I haven't been able to conclusively decide if the latest version of MT32-Pi has the same issue as MiniDexed and miniorgan or not at this stage.

[image]

rsta2, 2022-05-06:
Please add `usbignore=int3-0-0` to the file *cmdline.txt* on the SD card and try again. If this file does not exist, create it with this contents.

diyelectromusic, 2022-05-06:
That's done it! I don't believe it, I read about that update in the Circle history and hadn't connected it with the message in this log :)  Many thanks for that.  Why does that affect things?  I'm intrigued now as to why CircuitPython presents the int3-0-0 interface in the first place, so I might go off an do a bit more reading...

Anyway it also works on my MiniDexed too now.  Many thanks!

rsta2, 2022-05-06:
You are welcome. Good question, why it presents the `int3-0-0` interface. This is normally used by standard HID devices (keyboard, mouse, touchscreen, gamepad). The problem is, that when a device presents this interface, it has to offer a HID report descriptor, and when Circle tries to read this descriptor, it gets an error and the endpoint stalls. Further communication with this device is not possible then. With that `usbignore=` option Circle simply ignores the given interface, which is not needed for MIDI communication here.