rsta2/circle #294: Mouse failing setup on QEMU

zolbatar, 2022-05-07:
Using the circle LVGL class as a starting point, I'm trying to get mouse working on QEMU. I know the mouse pointer won't show but during init it fails on m_pMouseDevice->Setup(nWidth, nHeight), returning 0.

During boot, it picks up the QEMU USB keyboard and mouse.

It works fine with the same code on my real Pi.

Any ideas?

rsta2, 2022-05-08:
Yes, this `m_pMouseDevice->Setup()` call tries to initialize the mouse cursor, which is not supported by QEMU. LVGL will not work and does not initialize successfully on QEMU for this reason, this is known.

zolbatar, 2022-05-08:
I commented out the checks in mouse behaviour.cpp and added a lvgl mouse cursor and it works great, thanks for the help.

rsta2, 2022-07-29:
Because you have a solution for this problem I think this can be closed.

closed 2022-07-29 by rsta2