rsta2/circle #299: Boot from USB

nickverlinden, 2022-06-02:
Hi,

First of all, thanks Rene (neighbour, greetings from Belgium) for developing this framework. I've been developing a sampler for a bout two years now (https://hackaday.io/project/173554-opensampler), using Circle as a base. I progressed pretty far, more than you can see on the hackaday blog. Now, I want to boot from USB, so i can use the mmc to read sdcards that don't contain the operating system. Therefore i used the usb boot function as documented in the official Pi documentation. I'm using the serial bootloader and flashy to test, but am getting the following error after uploading the kernel image over serial:

logger: Circle 44.4.1 started on Raspberry Pi 4 Model B
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.68 emmc: Inquiry ACMD41 failed

Any ideas?

Kind regards,
Nick

rsta2, 2022-06-03:
There must be some call to `m_EMMC.Initialize()` somewhere in your code, which initializes the EMMC controller and tries to access the SD card. This call will fail, when there is no SD card inserted. You have to remove this call from the initialization and defer it to the place, when the SD card has been inserted by user interaction.

nickverlinden, 2022-06-03:
Of coarse that makes total sense! Thanks!

closed 2022-06-03 by nickverlinden