rsta2/circle #300: Multitasking

zolbatar, 2022-06-12:
I want to have a listener process running on the second CPU core, with core 0 doing regular round robin scheduling. I've successfully got CMultiCoreSupport running but I can't seem to get regular tasks running alongside on core 0. 

Does it support this? Or is it a case of multicore OR the regular CScheduler stuff?

rsta2, 2022-06-12:
The classes `CScheduler` and `CMultiCoreSupport` can be used at the same time. `CScheduler` allows to run multiple tasks on CPU core 0, but not on core 1, 2 or 3. This is a cooperative scheduler, so each active task must release the core from time to time, so that other tasks can run. This is normally done by calling `CScheduler::Get()->Yield()`.

`CMultiCoreSupport` allows to use three additional parallel threads of execution, which run on core 1, 2 and 3. `CMultiCoreSupport` only starts the secondary cores 1, 2 and 3, normally there is no further management necessary on these cores. More info on multi-core support in Circle can be found [here](https://github.com/rsta2/circle/blob/master/doc/multicore.txt).

zolbatar, 2022-06-12:
This answer is perfect and informs me I can do exactly what I need! I already have multiple tasks co-operatively scheduling and wanted to do some CPU intensive processing on the other cores.

I wanted to check it was possible before investing substantial time into it! Thank you.

zolbatar, 2022-06-12:
A final question (sorry to be a pain), does the code running LVGL stuff need to be running on core 0? I.e. can I update the GUI from one of the other threads?

rsta2, 2022-06-13:
I would recommend to call `CLVGL::Update()` from core 0. It's not clear, what will happen, when you call it from a secondary core.

closed 2022-07-29 by rsta2