rsta2/circle #302: The 20-tcpsimple sample won't connect to ethernet

jiri, 2022-06-14:
I've compiled and set up the tcpsimple sample. It builds fine and when I insert the SD, the RasPi ACT LED blinks five times signalizing that the code got booted properly. The ethernet line to my router gets opened (1G full duplex as expected), but DHCP is never attempted it seems. The LEDs on the ethernet port stay permanently on and don't blink.

I haven't changed the sample and USE_DHCP is defined, meaning the NetSubSystem should get initialized with the default constructor and use DHCP. Any idea what I might be doing wrong? I'm using a Raspberry Pi 4 purchased in 2018 or early 2019.

rsta2, 2022-06-14:
I guess you want to use your Raspberry Pi 4 without a connected display? In this case you have to enable the system option `SCREEN_HEADLESS` before build, otherwise the sample program will not work. Please read [this comment](https://github.com/rsta2/circle/blob/master/include/circle/sysconfig.h#L298) for more info.

The system option can be enabled by removing the comment tag in the given file, by adding the option `-d SCREEN_HEADLESS` to `configure` or by manually adding the following line to the file *Config.mk*:

```
DEFINE += -DSCREEN_HEADLESS
```

jiri, 2022-06-14:
Oh yeah, that totally worked :) I rarely ever use Raspberry Pis with a screen so it didn't even occur to me to check this. It might be worth mentioning in the getting started guide, I bet there's more people like me! Thank you for the blazing fast response and for the amazing work you do on this project.

closed 2022-06-14 by jiri