rsta2/circle #303: Freeze at USBHCI Initialize

nickverlinden, 2022-06-15:
I was trying to compile the touchscreen example because I was about to try a new usb based touchscreen for my project, but when running, it seems to get stuck here:

`
logger: Circle 44.4.1 started on Raspberry Pi 4 Model B
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.73 pcie: Link up, 5.0 Gbps x1
`

Sometimes I don't even get any logger output at all.
When i remove the part below from `boolean CKernel::Initialize (void)` it boots fine.

`
if (bOK)
{
bOK = m_USBHCI.Initialize ();
}
`

I recently changed my pi to boot of a USB drive. Perhaps it is related? I also tried removing everything from the usb ports before booting (also the boot drive itself right before i upload the kernel using flashy), but same result.

rsta2, 2022-06-15:
Yes, I guess this is caused by booting from USB. Have you tried USB earlier with Circle sample programs on this Raspberry Pi, before changing the boot device to USB? If so, did this work?

nickverlinden, 2022-06-15:
Correct, i did compile usb samples before on the same board and they were working. I also did a test a few months ago on the board reading files from a usb stick.

rsta2, 2022-06-15:
So there must be an incompatibility between the USB boot code and the Circle xHCI or PCIe host bridge driver. I put this on my TODO list, but unfortunately I have currently no idea, what the problem can be. Unfortunately there won't be a quick fix for this issue.

nickverlinden, 2022-06-15:
Ok, thx. Perhaps it is because the pi gpus bootcode already does some kind of initialization to be able to read the usb stick. I would offer to help and dig into it myself, but this is far beyond my abilities. For now, i'll change back to sd card booting. The user has to boot from the sd card, and after that can change the card if required. In the days of the mpc2000 (and other samplers/synths) they also first had to insert the boot floppy, so it's like a retro thing :-). Thanks for putting it on the list.

nickverlinden, 2022-06-16:
@rsta2 i'll have to do some more testing because it seems that for some reason i am unable to load anything on the pi anymore. It could have been just a coincidence that removing the USBHCI init code worked. I'm considering my pi may be fried somehow. I'll try to boot raspbian tonight to see if that still works.

nickverlinden, 2022-06-17:
@rsta2 good news, long story short it was a faulty power supply. It was an original raspberry pi 3a power supply, but it must have gone bad. I had a new one in box lying around, tried that and everything worked. Including booting from usb, and accessing usb devices.

closed 2022-06-17 by nickverlinden

rsta2, 2022-06-17:
Good to know. Thanks for info!