rsta2/circle #304: GPIO timing issues with multitasking

jiri, 2022-06-18:
I'm writing a program which (other that some other tasks like the syslog daemon) has 2 main tasks: an Art-Net server and an OSC server. The Art-Net server receives ArtDMX packets over a UDP socket and then bitbangs the values onto a WS2812b LED strip. This works without issues until I introduced another task, when the timings of the bitbanging get wonky and the LED strip stops working properly. This doesn't only happen with the OSC server task, it even happens if I just instantiate CTask directly (e.g. an empty task). Any idea what could be causing this?

Coincidentally, is there a more precise timing facility than nsDelay which isn't very precise in itself? The LED strip tolerances are +- 150ns and I still get uncomfortably close to those. I call EnterCritical() and LeaveCritical() around the transmission code, so interrupts shouldn't be the source of this issue.

From my understanding of how the multitasker works there's no reason a new task should interfere with how another task performs, so my intuition is that maybe creating another task somehow throws the system timer out of whack?

rsta2, 2022-06-18:
I'm afraid bitbanging the WS2812b LED strip is not a very good idea. Did you have a look at [this sample program](https://github.com/rsta2/circle/tree/master/addon/WS28XX/sample/simple)? It uses the SPI master to control the LED strip, which results in a more accurate timing.

jiri, 2022-06-18:
I did look it over, but I figured it could only run one LED strip through the SPI interface, no? I need to run way more than one strip, so I resorted to bitbanging since it always worked fine on Arduinos (but I get that those are way less complicated than the RPi, which I'm kind of misusing for this purpose).

Is my presumption about the SPI solution correct? I've also seen someone use the SMI interface to drive multiple strips at once, so I'll probably give that a go next, since I noticed that SMI is supported in Circle :)

closed 2022-06-18 by jiri