rsta2/circle #306: Flashing With Flashy Causes Transfer Error

elmfrain, 2022-07-09:
I've been trying **circle** for the past week and it works great!

So then I tried to enable the flashing functionality so I can more quickly upload my code the Raspberry Pi. I don't have a USB to Serial adapter, but I do have a HC-06 Bluetooth module. It basically does the same thing but wirelessly, and I made sure that the module is running at the 115200 baud rate. I can confirm that data is being transferred correctly.

So then I tried this:
* Added `FLASHBAUD = 115200`, `USEFLASHY = 1`, `SERIALPORT = /dev/rfcomm0` to `Config.mk
* Did `npm install` for Flashy
* Did `make all` in `/boot` circle directory, giving me the firmware and the `Kernel.img` of the bootloader.
* Put files into SD card of Raspberry PI (Zero) and booted it up.

I think I setup the flashing functionality correctly.

However, when I do `make flash` on my project, I get this:

```
node  ../circle/tools/flashy/flashy.js \
        /dev/rfcomm0 \
        --flashBaud:115200 \
        --userBaud:115200 \
        --reboot: \
         \
        kernel.hex

Opening /dev/rfcomm0 at 115200...ok
Sending reset command...ok
Waiting for device...ok
Fast mode enabled
Sending............................ok
Finished in 0.1 seconds.
Sending go command...

An error occurred during the transfer: format
make: *** [../circle/Rules.mk:223: flash] Error 9
```

Yeah, so am I doing something work here, or is it a bug?

rsta2, 2022-07-09:
Unfortunately the developer of Flashy is not available any more. Because it works all right with a normal USB-to-serial adapter here, I can only guess that this problem is caused by the specifics of your Bluetooth adapter (e.g. increased latency). Have you tried the default bootloader (without `USEFLASHY = 1`)?

elmfrain, 2022-07-09:
Yeah, I tried using the python flasher (without `USEFLASHY = 1`), but for some reason, it doesn't work on my machine because I get this:

```
python3 ../circle/tools/flasher.py kernel.hex /dev/rfcomm0 115200
ERROR: Serial Port /dev/rfcomm0 busy or inexistent!
```

Nevertheless though, I just received a USB to Serial adapter today (an FTDI breakout board), and it finally works!

```Opening /dev/ttyUSB0 at 115200...ok
Sending reset command...ok
Waiting for device...ok
Fast mode enabled
Sending.......................ok
Finished in 39.2 seconds.
Sending go command...ok
Closing serial port...ok
Done!
```

So my guess is that you are right! The Bluetooth module has too much latency, and Flashy wasn't designed for that.

Although, it wouldn't be a bad idea for the Bluetooth module to work with Flashy. It allows for remote development, and it can be convenient. Maybe something to work on in the future!

elmfrain, 2022-07-10:
Good News!

I've been tinkering with Flashy's code, and I've actually got the HC-06 Bluetooth to work! I don't know exactly how I got it to work, but I somewhat know. In simple terms, the flasher rams too much data at once which causes some data to get scrambled during transfer.

I'm planning to contribute my findings, but I want to iron out my research and code before I publicize it. I'd also want to know if it could work with the faster HC-05 Bluetooth module. The HC-05 has a max baud rate of 1382400, so this can be a great solution to wireless programming!

rsta2, 2022-08-24:
Some time has passed now and I want to ask, if you have been successful with your research work about using Flashy with the HC-05 Bluetooth module? Is there something, you can contribute?

elmfrain, 2022-08-30:
My apologies for the wait. Yes the HC-05 does work with Flashy flawlessly, however the HC-06 does need Flashy to work a little differently or else the data gets corrupted (perhaps it can't handle mass data). I'll publish my findings in a PR tomorrow from now.

rsta2, 2022-12-01:
Solved in Circle 45.

closed 2022-12-01 by rsta2