rsta2/circle #307: Device recommendation for I2S input

Buanderie, 2022-07-29:
Hello,

First, congratulations on providing such fine piece of software for all of us !

I would like to test I2S input (and ouput) on my Pi3 using Circle. I was wondering if there was any DAC/ADC chip and/or modules recommended for this purpose (o avoid ordering parts which would be incompatible with Circle). Preferably a cheap one (44100Hz would be largely enough), and also if possible, one with Line-IN AND Line-OUT ?

Thanks in advance :)

rsta2, 2022-07-29:
Thanks for appreciating Circle! Unfortunately I tested the I2S input only against the I2S output from a second Raspberry Pi running Circle. Perhaps @sebastienNEC can recommend an interface for I2S input, because the this code was based on a patch, provided by him?

sebastienNEC, 2022-07-29:
Hello, I was using back then the FePi but they don't seem to sell these anymore... But I guess any audio HAT would do. The hardest part is to configure the codec correctly over I2C, so I would look for codecs where there's a github project that did the configuration already, so that you may copy/adapt it :)

andrethomas, 2024-03-08:
@Buanderie did you find something that works?