rsta2/circle #309: Makeall fails under Windows with MINGW32 and all makefiles

ArnoldUK, 2022-08-09:
Followed instructions for Windows build using GNU make for Windows and gcc-arm-none-eabi toolchain.
At the MINGW32 bash prompt and in the git root folder for circle-master the command ./makeall -j is not a valid switch and ./makeall shows these errors:
$ ./makeall
MAKE Version 5.4  Copyright (c) 1987, 2009 CodeGear
Error makefile 6: Command syntax error
Error makefile 20: Command syntax error
Error makefile 29: Too many rules for target 'clean'
Error makefile 30: Command syntax error
*** 4 errors during make ***
I've download all the correct tools and the paths are all set correctly. Not sure if I'm at fault or the build for Windows process.

hpingel, 2022-08-09:
Did you try to use Git Bash for Windows when launching makeall?

ArnoldUK, 2022-08-10:
> Did you try to use Git Bash for Windows when launching makeall?

Yes, I followed the Windows build docs exactly and try to compile using Git Bash shortcut which launches git-bash.exe and a Mingw32 bash console. Seems that the make command is failing when calling each of the Makefile's with commands such as ifeq else ifneq end etc.. stuff that would normally run under Linux if you understand me.

ArnoldUK, 2022-08-10:
Here is the contents of my .bash_profile

`export "PATH=$PATH:/d/AVR-GCC/GnuWin32/bin"
export "PATH=$PATH:/d/AVR-GCC/gcc-arm/bin"
`
I have the required build tools install in a folder at D:\AVR-GCC
Does the export line look correct ? I copied it from the Windows build docs.

rsta2, 2022-08-10:
I guess your *make* tool is incompatible with the Circle make-files. Can you please try the version from [here](http://gnuwin32.sourceforge.net/packages/make.htm). You have to ensure, that this *make* is found first in the `PATH` environment variable.

ArnoldUK, 2022-08-10:
The version with Git Bash is 5.4 so Git Bash is using the incorrect make. I followed instrucitons here 'https://github.com/rsta2/circle/blob/master/doc/windows-build.txt'
and says to launch "Git Bash" so I have no idea how to change the make version to use.

ArnoldUK, 2022-08-10:
> I guess your _make_ tool is incompatible with the Circle make-files. Can you please try the version from [here](http://gnuwin32.sourceforge.net/packages/make.htm). You have to ensure, that this _make_ is found first in the `PATH` environment variable.

Ok, I've modified the first line in my .bash_profile for PATH to not use any paths currently set in my Windows environment to:
`export "PATH=/d/AVR-GCC/GnuWin32/bin"`
`export "PATH=$PATH:/d/AVR-GCC/gcc-arm/bin"'`

Appears I have some paths incorrectly set and all is working now and compiled without errors :)

ArnoldUK, 2022-08-10:
For anyone lese who has an issue with compiling under Windows using MINGW32 and gcc-arm-none-eabi toolchain, make sure the paths to these tools are first on the PATH environment variable. **rsta2** pointed this out to me and I was too lazy to see the mistake I made.