rsta2/circle #310: sample 15-files issue on RPI3

Evanok, 2022-08-17:
Hello,

I am currently using circle to run some benchmark tests on RPI with Baremetal env. This project is working great. Thanks for this great job. I would like to be able to read data from a file on a USB stick and write an output result on this USB stick.
I try to run the sample 15-files. my RPI circle OS is not able to detect the USB stick device.

[image]

I try to check the doc and README file and everything seems to be fine.

I check my sysconfig file:

```
[arthur * remove_neon_fft_support] cat include/circle/sysconfig.h  | grep -i USE_USB_SOF | grep define
#define USE_USB_SOF_INTR
```

My usb stick is formatted with FAT32 partition type as requested. I try to use all the USB port on the RPI.
I can see in USB wiki that some RPI can have some stability issues. Did I miss something? An option to add to my cmdline?

My current boot cmdline:

```
[arthur * remove_neon_fft_support] cat boot/cmdline.txt
console=serial0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
```

Thanks,
Arthur.

rsta2, 2022-08-17:
Thanks for appreciating Circle. Please add the option `usbpowerdelay=1500` to your file *cmdline.txt* on the SD card and try again. Some USB devices need more time than the default 510 ms from powering on to the first access. This options sets the delay to 1500 ms.

Evanok, 2022-08-17:
Thanks for the quick reply. Unfortunately, I have the same result.

My USB stick information: 

[image]

My new cmdline.txt in my sd card:

```
lambe@DESKTOP-AVJV044 MINGW64 XXXX (master)
$ cat /E/cmdline.txt
console=serial0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles usbpowerdelay=1500
```

Is it supposed to work on any USB port?
Do I have to try to add more delay?

rsta2, 2022-08-18:
That's odd. For explanation according to your Circle log above the problem is, that your USB flash drive is not detected at all. This should work at any USB port of the Raspberry Pi 3 B+. There must be an USB interface `int8-6-50` displayed to be found somewhere in the log, or at least some error message, that the device or configuration descriptor cannot be read from an USB device. But both does not happen according to your log.

So your flash drive is not detected at all, which is a hardware thing. The partition layout of the drive does not count here. There was some chance, that it was the power on delay, which causes the problem, but 1500 ms should be enough for a normal USB flash drive.

Can you try an other USB flash drive with Circle? Another option would be to read/write to the SD card instead. Please see the sample program in *addon/SDCard/sample*, if you want to try this.

BTW. the command line in *cmdline.txt* can normally be empty for default Circle operation. For this case the `usbpowerdelay=1500` option would be the only one in this file. The additional options in your command line have an effect for Linux only, but should also not disturb Circle operation.

Evanok, 2022-08-18:
I have another USB key available for the test but I need to do a backup first. There are a lot of data on it. It is going to take some time.. It is a good opportunity to do a backup. It is always useful. I just remove all the useless options in the cmdline.txt.  I just keep the USB delay option and the result is better!

I can see: 

```
unsd: Capacity is 29510 MByte
partm: Drive has no supported partition
```

Perhaps all my useless option was hiding the USB delay option. Perhaps all my option string was too long for parsing? At least now I have a real error. So probably that there is an issue with my USB partition format. I format my USB stick on Ubuntu with Gparted. Nothing fancy.

I just run a test with the SDcard sample and it is working perfectly. This alternative solution will do the trick for my test.
I don't need to fix the USB issue anymore. 

Do you want me to finish the test/investigation about USB? if you are not interested I can just close this issue. It will be just great to add in the README USB sample file that the user must add the usbdelay option in case of trouble to detect USB device.

rsta2, 2022-08-18:
Thanks for testing! There may be a problem in the *cmdline.txt* options parser, so that it ignored the `usbpowerdelay=1500` option. I have to check this. I will add a note about this option to the sample's README file.

There is another sample program, you could try with your USB flash drive. Circle comes with two different FAT file system implementations, a very limited one, which is used in *sample/15-files* and a full implementation, which is based on the external FatFs file system library. Please have a look to *addon/fatfs/sample/README* for details. You have to modify the file *kernel.cpp* to access an USB flash drive (default is SD card). You still may get the "partm: Drive has no supported partition" message, but perhaps it will work with FatFs any way. The `usbpowerdelay=1500` option is still necessary.

closed 2022-08-19 by Evanok

rsta2, 2022-08-19:
I added a note on `usbpowerdelay=` to the sample's README file. I also checked the command line option parser. It seems to be all right.