rsta2/circle #313: Pi Zero 2 W: Different revisions, different WLAN firmwares

hpingel, 2022-09-02:
Hi Rene,
Yesterday I noticed that two different PI Zero 2 W PCBs can have different WLAN hardware: These logger lines exist on different Pis:

`wlan: ether4330: chip 43430 rev2 type 1` (my Pi)
`wlan:  ether4330: chip 43430 rev1 type 1` (the Pi of a colleague)
In the next line my colleague gets this:
`wlan: ether 4330: can't enable HT clock: csr 50
wlan: I/O error`

Now, I was wondering if this error could be related to the WLAN firmware files. But it could also be something else like a power supply or a hardware fault.

Today I came accross this ticket: https://github.com/RPi-Distro/firmware-nonfree/issues/23

I tried to understand if this is of any relevance to the issue above and am not convinced. But anyway, I wanted to let you know that there are some firmware files for the Zero 2 W in https://github.com/RPi-Distro/firmware-nonfree/tree/bullseye/debian/config/brcm80211/brcm
that are not yet grabbed by wget from Circle's side.

The missing files are:
brcmfmac43436s-sdio.bin
brcmfmac43436s-sdio.txt

And there were some changes to some files after October 2021.

Cheers,
Henning

hpingel, 2022-09-02:
I know grasped that a firmware brcmfmac43436s-sdio.* will only be necessary for a future revision of the Pi Zero 2 W that is probably not yet available due to the global chip crisis anyway.

I just tested all Pi Zero 2 W boards I have here and they all have either rev1 or rev2.

rsta2, 2022-09-03:
The WLAN device driver has [separate entries](https://github.com/rsta2/circle/blob/master/addon/wlan/ether4330.c#L294) for revision 1 and 2 of the 43430 chip, so this difference should be handled well. As I understand it, there is currently no need to change something. But I will keep an eye on this and this issue will remain open. Thanks for this info!

rsta2, 2025-07-29:
The updated WLAN firmware (see #602) has a specific driver for newer RPi Zero 2 W. Unfortunately I don't know the chip ID and revision for this firmware, because I do not have such RPi Zero 2 W. If somebody notes, that his/her WLAN does not work with a RPi Zero 2 W, he/she may add the IDs [here](https://github.com/rsta2/circle/blob/7d7eabde2b9e8cdff5c169643a80d5866714a780/addon/wlan/ether4330.c#L299). They can be found in the log messages.