rsta2/circle #314: Touchscreen support for Waveshare 7.9 Inch Screen

nickverlinden, 2022-09-08:
Hi Rene,

This display:
[https://www.waveshare.com/7.9inch-hdmi-lcd.htm](https://www.waveshare.com/7.9inch-hdmi-lcd.htm)

I added some code to usbdevicefactory.cpp to add touchscreen support for this display. It uses the regular HID driver, but in the current circle implementation, it is not detected. Adding this piece of code enables it to be used. I was just wondering if this is the right way to do it, I read the usb doc, but i don't fully understand all the device codes and their meaning. I don't know if every device identifying as 'int3-0-2' should be considered a GenericHIDDevice, that's why i'm checking the vendor id. Is this the way to do it, or can we safely assume 'int3-0-2' is always a touchscreen without comparing the vendor id?

usbdevicefactory.cpp, line 105:
`else if (pName->Compare ("int3-0-2") == 0)
	{
		CString *pVendor = pParent->GetDevice ()->GetName (DeviceNameVendor);
		assert (pVendor != 0);

		if (pVendor->Compare ("ven712-9") == 0)	// Waveshare 7.9 Inch Touchscreen
		{
			pResult = GetGenericHIDDevice (pParent);
		}

		delete pVendor;
}`

Kind regards,
Nick

rsta2, 2022-09-09:
I generally added the interface `int3-0-2` for generic USB HID devices on the *develop* branch to be part of the next release. As I understand the USB HID 1.11 specification, this is not a standard interface, but adding should not cause problems with other devices and maybe there are other devices out there, which use this interface. Thanks for reporting this issue!

nickverlinden, 2022-09-10:
Perfect, thanks!

closed 2022-09-10 by nickverlinden