rsta2/circle #316: USB MIDI support only supports BULK transfer endpoints

diyelectromusic, 2022-09-23:
It would appear that there are several common USB MIDI controllers (e.g. those by Novation) that seem to present an INTERRUPT endpoint rather than a BULK endpoint.  This means that they are ignored in usbmidi.cpp due to the check for just BULK endpoints.

It would appear that if this is changed to BULK or INTERRUPT then these controllers will work, but not really understanding how USB works, I don't know the implications of this change...

For more discussion and some details, see this thread as part of MiniDexed: https://github.com/probonopd/MiniDexed/discussions/258

Kevin

rsta2, 2022-09-23:
According to the [USB MIDI class spec. 1.0](https://usb.org/sites/default/files/midi10.pdf) only bulk endpoints are used for MIDI streaming. If these devices use interrupt endpoints, this is not class conform. Nevertheless, if these devices work with standard operating systems, there should be a way to get them working with Circle too. I suppose that the test for bulk endpoints in the USB MIDI driver can be modified to except interrupt endpoints too. I do not think, that this will cause problems elsewhere. I will modify the driver tomorrow.

diyelectromusic, 2022-09-23:
I think a later version of the spec allowed support for interrupt endpoints, but as I say, I'm not really a USB person!  See: https://www.usb.org/sites/default/files/USB%20MIDI%20v2_0.pdf

rsta2, 2022-09-24:
Yes, but according to the "Class-specific AC Interface Descriptor" (see B.3.2) in the USB MIDI class 1.0 spec. both given devices support the revision 1.0 of the class spec. The field `bcdADC` is 0x100. I don't think the USB MIDI driver in Circle does work with devices at all, which are compatible with the 2.0 spec. This is not supported.

rsta2, 2022-09-24:
The USB MIDI driver has been updated on the *develop* branch to allow to use Interrupt EPs.

This driver may work with USB MIDI controllers, which support the USB MIDI Class 2.0 spec., because these devices should support a special mode for 1.0 hosts. But this has not been tested by myself, because I do not own such a USB MIDI controller. My previous comment regarding USB MIDI 2.0 support may be wrong.

diyelectromusic, 2022-09-24:
> The USB MIDI driver has been updated on the _develop_ branch to allow to use Interrupt EPs.

Well as I say, I don't really know USB, so I guess we'll have to see how people get on! :)

Many thanks - will take a look.

Kevin

diyelectromusic, 2022-09-24:
> Yes, but according to the "Class-specific AC Interface Descriptor" (see B.3.2) in the USB MIDI class 1.0 spec. both given devices support the revision 1.0 of the class spec. The field `bcdADC` is [...]

Ah yes, I see what you mean.  So the Novation ought to have bcdADC to 0x200 if it wants to use INTERRUPT endpoints... Hmmm. :)

pik33, 2022-09-25:
The Novation Impulse 61 has  bcdADC = 1.00 and then  Transfer Type = Interrupt. Doesn't work in MiniDexed before, the updated experimental build now works.

rsta2, 2022-09-25:
@pik33 Thanks for info!

rsta2, 2022-12-01:
This issue has been fixed in Circle 45. @diyelectromusic thank you for finding the problem.

closed 2022-12-01 by rsta2