rsta2/circle #320: Waveshare 7.9inch DSI version compatibility

nickverlinden, 2022-10-17:
Hi!

Right now i'm usig the Waveshare 7.9inch wide display using HDMI and Usb for touch. I recently discovered there is a DSI version of this display with i2c touchscreen, I would prefer to use that becuse now I have an thick usb cable sticking out the back of my enclosure going into one of the pi's usb ports. 
It seems in linux you seem to require a special driver for this screen. I dont know much about DSI except that it's a closed standard, and always assumed the display resolution was fixed to 800x480. This screen has a much larger resolution. Does anybody know if the dsi version will work in circle?

[https://www.waveshare.com/7.9inch-dsi-lcd.htm](https://www.waveshare.com/7.9inch-dsi-lcd.htm)

Kind regards,
Nick

rsta2, 2022-10-17:
I did not test it, but unfortunately I'm afraid this display will not work with Circle. It requires special drivers for Linux, which are not available for Circle. Even if somebody would want to develop such drivers for Circle, this seems to be impossible, because the source code of the Linux drivers is not available for reference.

nickverlinden, 2022-10-17:
@rsta2 Thanks, that what i thought. As i understand it every mipi display is different, so the default framebuffer driver inside the gpu probably does not natively support it. I think most of the existing after market DSI displays just reverse engineered the official pi touchscreen's communications and are all using the exact same resolution.

closed 2022-10-17 by nickverlinden