rsta2/circle #328: SQLite3 example?

froggestspirit, 2022-11-27:
I was attempting to compile sqlite3 for use with circle, though has no luck. I'm not sure if this might be something useful for others if it were to be supported?

rsta2, 2022-11-28:
Have you tried to use the [circle-stdlib project](https://github.com/smuehlst/circle-stdlib) for the build? It provides C-standard-lib support for Circle and it's more likely to work for porting projects, which are targeted for UNIX/Linux environments.

rsta2, 2022-11-29:
I did a quick hack and some functions of SQLite3 already work (in QEMU):

[image]

If you want to try it, extract the attached archive into the directory *samples/06-sqlite/* of the cloned circle-stdlib project. Put the files from [this SQLite3 source archive](https://sqlite.org/hp1.html) into the same directory and apply the patches in *sqlite.diff*. After configuring circle-stdlib for example with `configure -r 3 -p aarch64-none-elf- --no-cpp` and building it with `make -j`, go to the *samples/06-sqlite/* directory and enter `make -j`. You will get some warnings, which are not critical.

*func.c* contains the stubs for functions, which are not implemented by circle-stdlib. Most functions in this file are not implemented yet. Saving a database to a file does not work yet.

[06-sqlite.zip](https://github.com/rsta2/circle/files/10112295/06-sqlite.zip)

froggestspirit, 2022-11-29:
Thank you! I'll give that a try this afternoon

rsta2, 2022-12-04:
@froggestspirit Did this work for you?

froggestspirit, 2022-12-04:
Yes, the example you posted worked! I was working on trying to get file support working with the sd card, but haven't found the time to get back to that yet.

froggestspirit, 2022-12-11:
I spent some time trying to get some file reading/writing working, but the closest I got was porting a SQL logger for arduino (which doesn't really support querying). I'm thinking that this needs a custom VFS? I grabbed the test_demovfs from the sqlite site, and tried adjusting it to use the fopen, fread, fwrite, etc. I've attached what I got here, but most of what I tried results in the SQLITE_CANTOPEN (error code 14).
[06-sqlite.zip](https://github.com/rsta2/circle/files/10203331/06-sqlite.zip)

rsta2, 2022-12-12:
I also did some debugging with QEMU here. I think the problem has to do with the `stat()` function, which always returns an error. I guess, it is not supported by *circle-newlib*. Also the file names are a problem, because they need the prefix `"SD:"` to access the SD card, which is not UNIX standard. SQLite cannot handle this. Unfortunately I'm not familiar with SQLite and it takes time to understand, what it does. Perhaps I will have another look at this after Xmas when my current project has reached a reasonable state.

rsta2, 2022-12-22:
I already got most features running. You can have a look on it [here](https://github.com/rsta2/sqlite). The main problem were the functions `stat()`, `fstat()` and `fcntl()`, which exist in the circle-newlib project, but are only stubs, which always return an error. I replaced these functions with own versions in *func.c*. `fcntl()` cannot be really implemented for record locking, but always returns success now.

Unfortunately there is still a problem. Under some conditions SQLite reports a "malformed schema on disk" after creating a database. This does not happen, when creating the database in memory and writing it to disk with `.save` then. This problem has still to be analyzed.

froggestspirit, 2022-12-22:
Thank you! I have some time off over the holidays, I should be able to test it out

rsta2, 2022-12-22:
You are welcome. In the meantime I have found the "malformed schema" problem. `fsync()` needs to be implemented. I did this in a quick hack and the problem was gone. For a real solution we'll need more time and modifications in circle-newlib. I will let you know, when it's fixed.

rsta2, 2022-12-28:
The project has been updated. Thanks to @smuehlst there is additional support from the circle-stdlib project now. `fsync()` is implemented. I did some tests and currently I'm not aware of any problems. It is currently not clear, if SQLite opens the same file multiple times with different file descriptors. This is not supported by FatFs and could cause problems. I still have to check, if this happens.

rsta2, 2022-12-29:
@froggestspirit My impression is, that everything, which can be supported, is working now. Also SQLite does not open the same file multiple times. BUT, there are open questions on the internal function of SQLite and together with circle-stdlib and the FatFs file system module. It is not clear, why SQLite does not work, when `fsync()` is not implemented on a single user system like this one, and why it does not work, when `fsync()` is called inside `fstat()` to update the file information, before it is fetched to be returned in the `struct stat`.

Again, SQLite seems to work in the current state of the repo [sqlite](https://github.com/rsta2/sqlite), but it would possibly take much more time to answer the given questions. Unfortunately we don't have the time to analyze this further and can only present this current state. It is currently not planned to develop this further, and it is likely that the repo *sqlite* will be removed from my project space in some weeks.

froggestspirit, 2022-12-29:
Thanks! I just downloaded a zip of the repo. I should have some more time to look at it soon

smuehlst, 2022-12-29:
@froggestspirit If you can find out more about the strange behavior regarding `fsync()` in SQLite I would be interested. I currently don't have time to investigate this myself, but if there are insights that help improve the FatFS-based file operations in circle-stdlib they are certainly welcome.

froggestspirit, 2023-01-02:
I did a quick test with this, instead of using the CLI (since I'm aiming to use this for a sort of mp3 player), I ran some commands to create a table, insert a column, then read what's in that table, and re-insert the same records (multiplied by 3). It looks like it's working!

rsta2, 2023-02-22:
OK, I suppose, it is working. So this can be closed. The [sqlite repo](https://github.com/rsta2/sqlite) will remain as an example and for testing purposes.

closed 2023-02-22 by rsta2