rsta2/circle #329: Expanded support for CP210x family USB serial devices

dwhinham, 2022-12-01:
This issue was reported by a user with a CP2104 device (following on from https://github.com/dwhinham/mt32-pi/issues/305).

Removing a part number check in the CP2102 driver [(relevant code)](https://github.com/rsta2/circle/blob/182ae1cf6d43016393b3ca6a3932f26d5417ee0e/lib/usb/usbserialcp2102.cpp#L93-L98) apparently gets it working for CP2104, and possibly other members of the CP210x family as they appear to be very similar.

[This PDF](https://www.silabs.com/documents/public/application-notes/an976-cp2102-3-4-9-to-cp2102n-porting-guide.pdf) (page 2) suggests that the differences are very minor (e.g. number of stop bits supported).

Searching for `partnum` [within the Linux driver](https://elixir.bootlin.com/linux/latest/source/drivers/usb/serial/cp210x.c#L2007) also reveals where behaviour varies depending on the part number.

It would probably make sense to:
- Rename the driver to CP210x.
- EITHER: Remove the part number check and deal with any further issues as reported by users of the hardware (easiest solution).
- OR: Expand the part number check to other known part numbers and adapt the behaviour of the driver to enforce the correct maximum baud rates, stop bits etc in a similar way to the Linux driver to account for the minor differences in the CP210x family (requires more effort/testing with the hardware).

I would offer to help, but sadly I don't possess any of the devices concerned, hence I'm only able to ensure the issue is tracked.

Many thanks as always!

rsta2, 2022-12-02:
@dwhinham Thank you for the comprehensive information as always! I will try to implement the second option. I'll need a few days.

rsta2, 2022-12-02:
There is a modified CP210x driver on the *develop* branch. This went quicker than expected. I tested it with a CP2102. Please let me know, if more modifications are necessary. I'm not sure, if we should add more vendor/device ID combinations. The Linux driver has a lot of them. I wouldn't include them all, but perhaps the missing SiLabs factory defaults should be added (0xEA61, 0xEA70).

dwhinham, 2022-12-02:
Perfect solution, thanks Rene. I personally can't think of any further changes necessary, and the way you have implemented this makes it very easy to extend later on if need be.

Cheers!

rsta2, 2022-12-03:
OK, thanks. Then we should wait, if somebody wants another modification in the coming time. Cheers!

rsta2, 2023-05-22:
This feature is in Circle 45.2 now.

closed 2023-05-22 by rsta2