rsta2/circle #330: Best way to get millis() back

goran-mahovlic, 2022-12-03:
Hi, thank you on such a great project!

I am trying to add eez-framework that is needed for eez LVGL GUI builder with flow support so we can also do active stuff from builder, and I got almost everything working but not sure what would be best way to get millis from your system. I am currently using CTimer::Get ()->GetTicks (); and I do see some activity - but I know this is no way to go...

closed 2022-12-03 by goran-mahovlic

rsta2, 2022-12-03:
Yes, this is the way to provide `millis()`. Please note, that the returned value may wrap and start from 0 again. Thanks for appreciating Circle!

goran-mahovlic, 2022-12-03:
Thank you so much! 

I see that LVGL is using the same so it may be good enough.

https://youtu.be/jHp4GPk0dVA

I am still not good at GUI builder but this is how this demo looks. 

It is without any additional code - flow is doing all activity

[image]

One more question can I remove cursor at the startup - or maybe have even faster start :)

rsta2, 2022-12-04:
Looks great. Startup can be made even faster with these entries in the file *config.txt* on the SD card:

```
boot_delay=0
disable_splash=1
```

If you add the following line to your *Config.mk* file you can reduce it by another second on a Raspberry Pi 1:

```
DEFINE += -DNO_CALIBRATE_DELAY
```

There is also a WiFi sample in *addon/wlan/sample/hello_wlan/*, but unfortunately it does not work on the RPi 1, which has no WiFi.