rsta2/circle #339: CLVGL issues with WaveShare 5" Touchscreen

daparker26, 2022-12-26:
First, let me say that Circle has been a joy to work with as someone who is new to bare metal development. Thank you for sharing and maintaining it.

I am using a WaveShare 5" resistive touch screen. Circle detects it and it works fine with the touchscreen calibrator and the touchscreen sample:

usbdev1-3: Device veneef-5 found (FS)
usbdev1-3: Product: WaveShare WS170120
usbdev1-3: Interface int3-0-0 found
usbdev1-3: Using device/interface int3-0-0

However, when I use it with CLVGL, whether it is my own application or the sample, touch events are not working. The log gives me this:

lvgl: [Warn]	(480.726, +30)	 indev_pointer_proc: X is 2976 which is greater than hor. res 	(in lv_indev.c line #362)
lvgl: [Warn]	(480.726, +0)	 indev_pointer_proc: Y is 549 which is greater than ver. res 	(in lv_indev.c line #368)

The touchscreen resolution is 800x480, so I assume the touch coordinates are not aligned with the resolution. I'm using step 45 on a Raspberry Pi B2.

rsta2, 2022-12-27:
Thank you for appreciating Circle. From the error messages from LVGL I guess, that the touchscreen configuration has not been set or not set right in the file *cmdline.txt*. Maybe a typo or something like that occurred. The touchscreen calibrator writes some info on the screen, when it completes. It looks like that:

```
touchscreen=133,3998,280,3977
```

This option has to be written manually into the first line of the file *cmdline.txt* on the SD card with an editor. If this option is not set accordingly, the touchscreen reports invalid coordinates, which lead to these messages.

daparker26, 2022-12-27:
Ohhh, you are right. I had the touchscreen configuration in config.txt instead of cmdline.txt. I appreciate your response!

closed 2022-12-27 by daparker26