rsta2/circle #342: Backport TLCDRotation Rotation, TLCDMirror Mirror

probonopd, 2023-01-14:
In `ssd1306device.cpp`, it would be nice to have TLCDRotation Rotation and TLCDMirror Mirror like in https://github.com/dwhinham/mt32-pi/blob/master/src/lcd/drivers/ssd1306.cpp.

This should allow us to, e.g., rotate the display by 180 degrees as [requested](https://github.com/probonopd/MiniDexed/discussions/412) by one of the MiniDexed users.

rsta2, 2023-01-15:
The SSD1306 driver has been ported by @diyelectromusic. I don't know, if he has time to do this?

diyelectromusic, 2023-01-15:
I'll try to take a look and see.  What I wasn't sure about is what to do about the HD44780 side - do we introduce an API to the "character device" to set rotation but just ignore it for the HD44780 and just implement it for the SSD1306 do you think?

(I've not had a proper look yet, but I think it is just a matter of setting the correct initialisation parameters for the SSD1306 when instantiating the display object)

Kevin

rsta2, 2023-01-16:
Kevin, I think your suggestion is good. Another possibility would be an additional rotation parameter to the constructor of the SSD1306 class with the default value 0. Please chose the solution, you prefer. Thanks.

rsta2, 2023-01-17:
This feature is implemented on the *develop* branch now.

closed 2023-05-22 by rsta2