rsta2/circle #344: add ability to clear raw keyboard handler

incanus, 2023-01-17:
Thanks as ever @rsta2. What do you think about adding the ability to clear the raw keyboard handler? I have been using this for some time on my own project and am wondering why it doesn't exist in the library. 

```c
void RegisterKeyStatusHandlerRaw (TKeyStatusHandlerRaw *pKeyStatusHandlerRaw,
                                  boolean bMixedMode = FALSE);

void ClearKeyStatusHandlerRaw() { m_pKeyStatusHandlerRaw = 0; }
```

rsta2, 2023-01-18:
I added a similar method `UnregisterKeyStatusHandlerRaw()` on the *develop* branch. The term "unregister" has been already used somewhere else in Circle, so for being homogeneous. Thanks for the hint!

rsta2, 2023-05-22:
This feature is in Circle 45.2 now.

closed 2023-05-22 by rsta2