rsta2/circle #345: CUSBHCIDevice hangs after chain boot

Reesy1979, 2023-01-19:
I have multiple circle kernels on the SD card, I created a kernel to allow a user to select the kernel to load.  I then use the chainboot functionality to load that kernel.  This works fine unless the new kernel being loaded also uses CUSBHCIDevice, if it does it hangs while performing a blocking call - waitForFrame, the value from register DWHCI_HOST_FRM_NUM never changes.

void CDWHCIFrameSchedulerNoSplit::WaitForFrame (void)
{
	CDWHCIRegister FrameNumber (DWHCI_HOST_FRM_NUM);
	m_usNextFrame = (DWHCI_HOST_FRM_NUM_NUMBER (FrameNumber.Read ())+1) & DWHCI_MAX_FRAME_NUMBER;

	if (!m_bIsPeriodic)
	{
		while ((DWHCI_HOST_FRM_NUM_NUMBER (FrameNumber.Read ()) & DWHCI_MAX_FRAME_NUMBER) != m_usNextFrame)
		{
			// do nothing
			
		}
	}
}

I can re-produce the hang within a single kernel, ie create the CUSBHCIDevice, destroy it and then create it again - it will hang in the same place as it does after the chainboot.

The Chainboot example does state that some devices do not get reset correctly, I assume CUSBHCIDevice is one of those.

"* Some devices are not reset in the destructors of their driver classes.
  This may cause problems with some devices. Please report such issues.
  The devices used in this sample should work in any way."

Does anyone has any unpublished code to close down CUSBHCIDevice cleanly?

rsta2, 2023-01-19:
The destructor of the class `CUSBHCIDevice` should be properly implemented, otherwise the *sample/38-bootloader* would not work. I did several tests here with a Raspberry Pi 2B with the bootloader sample and also with a test program, which creates, initializes and destroys an instance of `CUSBHCIDevice` multiple times. Everything worked well. See the test program attached.

With which RPi model are you working and in which mode (32 or 64 bit)? Do you have special system options enabled or other special configurations?

[usbtest.zip](https://github.com/rsta2/circle/files/10460752/usbtest.zip)

Reesy1979, 2023-01-19:
Thank you for confirming it works, no doubt I've messed something up along the way.
I'm using a PI Zero 1 32bit mode - I forgot to mention that I'm using circle via the circle-stdlib project so perhaps that is the difference.  I'll repeat some tests using circle on its own and see if I get the same results as you.

Thanks again for your help and for your time creating this great software.

closed 2023-01-19 by Reesy1979

Reesy1979, 2023-01-19:
Issue appears to be with my USB device, I was using a retroflag GPI case - code was detecting device ven45e-28e.  Using your test code I got the same issue, second attempt to connect hangs.  I took my Pi Zero out of the retroflag case and used a normal PS4 controller and everything works correctly.  Just thought I would mention it in case anyone else has issues.