rsta2/circle #346: USB keyboard translation behaviour does not handle 'ctrl' modifier correctly

kevinboone, 2023-01-23:
If I understand correctly, CUSBKeyboardDevice has a CKeyboardBehaviour which has a CKeyMap. CKeyMap maps scan codes to sequences of characters that correspond to ANSI terminal codes. In CKeyMap the 'ctrl' key is handled correctly for ctrl+letter combinations, but there's no support for ctrl+home, ctrl+function , ctrl+arrow, etc. Some of these, particularly the ctrl+arrow keys, are very widely used. 

It would be nice if this support could be added. However, it looks rather complicated. I think that the scan code is converted into a value from the TSpecialKey enum, and then this is converted into a character sequence. So everything that can generate a character sequence needs to have its own entry in TSpecialKey. 

So I think what would need to be done is as follows:

- Add all the KeyCtrlLeft, KeyCtrlRight, KeyCtrlHome, KeyCtrlF1, etc., values to TSpecialKey 
- Add an additional column to each of the mapping tables in keymap_XX.h, for the ctrl modifier, filling in all the appropriate values. Fortunately, I think the values in the additional column would be the same in all the maps that are currently available 
- Modify the logic in CKeyMap::Translate to detect the 'ctrl' modifier, and select the appropriate column of the lookup table in the appropriate map
- Modify CKeyMap::s_KeyStrings to include the character sequences for each of the new KeyCtrlLeft, etc., values. I think this array needs to be ordered the same as the definition of the TSpecialKey enum entries.

This looks like a lot of work. I wouldn't mind doing it myself but, given the amount of effort involved, I'd like at least to be sure that I have understood how the key translation actually works.

Unfortunately, because I'm using circle-stdlib, I don't have a way simply to get raw scan codes and translate them in my application code (so far as I know).

kevinboone, 2023-01-23:
I've just realized that, in issue #169, somebody said he already had a patch for this. I don't know if it was ever integrated, however.

kevinboone, 2023-01-24:
Never mind -- I made the changes and I've logged a PR.

rsta2, 2023-05-22:
The Ctrl+navigation keys are supported in Circle 45.2 now.

closed 2023-05-22 by rsta2