rsta2/circle #348: A fully-synchronous filesystem mode might be useful

kevinboone, 2023-02-06:
I note that there is a method void `CFATFileSystem::Synchronize()` which is documented to flush file buffers. It isn't entirely clear to me what the consequences are of _not_ calling this method. That is, I don't know when (if ever) the disk/SD will be made fully up-to-date.

It would be nice if there were a way to ensure that there were never a need to call this method. That is, it would be good if we could be sure that, after any file write (or directory creation, or rename), data on disk/SD could be assumed to be up-to-date by the time the method returns. This would be like the 'sync' mount option on Linux. Of course, potentially this would make disk/SD throughput slower, but this could be a necessary sacrifice.

In remote-sensing and battery-powered applications, a proper shut-down cannot be guaranteed. While I fully understand that there's really no way to protect a filesystem against a power failure whilst writing a large file, it would be good if this did not leave a broken filesystem.

To be honest, I don't know whether any change is necessary -- it might be that disk access is already sufficiently synchronous that I don't need to worry about shutting down. I have so far not been able to break a filesystem by killing the power whilst writing a file. I just don't know if that's by design, of I've just been lucky.

rsta2, 2023-02-06:
There are two different implementations of file system drivers for the FAT file system in Circle. The historic first in *lib/fs/fat/*, which has limited function and the ported FatFs library in *addon/fatfs/* with full function and higher performance.

The first one uses a buffer cache, which is flushed by calling `CFATFileSystem::Synchronize()` or automatically, when a file is closed, which was open for write.

I suppose you are using circle-stdlib/-newlib, which is based on the second option (FatFs library). As I understand it, with FatFs you can power off your Raspberry Pi at any time, when no file is open for write. You do not need to explicitly un-mount the file system before. Furthermore, when a power off occurs, when a file is open for write, you can loose the contents of this file at maximum, not more. You can improve this situation even more, when you call `fsync()` from time to time for this file. `fsync()` is implemented in circle-newlib and calls the FatFs function [f_sync()](http://elm-chan.org/fsw/ff/doc/sync.html) by itself. Please read this [application note](http://elm-chan.org/fsw/ff/doc/appnote.html#critical) for info about possible data loss with FatFs.

Currently it seems impossible to me to modify one of these file system drivers for a fully synchronous operation, but perhaps the situation with FatFs is enough for you.

kevinboone, 2023-02-07:
Thank you for the prompt reply. I think the situation is perhaps worse with circle-stdlib than it is with Circle itself. If you're programming with Circle, then you know you're working outside an operating system. You know that you probably have to take actions in code that you would not take, if you were working on an operating system. In any case, you're most likely working with code specifically written for that environment.

Using circle-stdlib obscures the fact that there is no operating system, which is both its strength and its weakness. circle-stdlib makes it easy to port existing code and libraries that were originally intended for Unix-like systems, without having to make many changes. This is great, but can lead to problems.

On Linux, a background process flushes disk changes periodically. Also, modern Linux filesystems are journalling, so it's relatively easy to repair changes related to a bad shut-down. Without these protections, the developer has to take a lot more care. But if you're using code in existing libraries, it's not always easy to work out what changes need to be made.

In any event, if circle-stdlib uses ChaN's FATFS implementation directly, and not Circle's filesystem support, then this isn't a problem that can be solved in Circle itself. So please close this.

By the way -- is there a place to ask general questions about Circle, without having to raise an issue in GitHub? I have many questions ;)

rsta2, 2023-02-07:
Understood. There is a Q&A category in the [Discussions](https://github.com/rsta2/circle/discussions/categories/q-a) for asking questions without open an issue.

closed 2023-02-07 by rsta2