rsta2/circle #359: compiler issues with conflicting types when using C stdio.h

ArnoldUK, 2023-04-13:
I'm having compile errors with many conflicting declarations in /usr/include/newlib/stdio.h
error: `typedef __int_least64_t time_t` and `typedef _TIME_T_ time_t`
note: `previous declaration as typedef long in time_t` with `typedef signed long time_t`

This happens when compiling any C code that uses the stdio.h file along with following includes in kernel.h:
#include <circle/logger.h>
#include <SDCard/emmc.h>

Working on Debian with all the build essential tools and arm-none-eabi compilers. I can compile the
full circle SDK and all samples fine apart from any C code using the stdio C libraries.

Is there a compiler flag that I should be using when compiling using C99 standard and using stdlib ?

rsta2, 2023-04-13:
Please use one of the [recommended toolchains](https://github.com/smuehlst/circle-stdlib/blob/master/README.md?plain=1#L23). They only need download and archive extraction, no separate install process. With this toolchain I cannot reproduce the problem here. Please ask further questions on the circle-stdlib project [here](https://github.com/smuehlst/circle-stdlib/issues).

ArnoldUK, 2023-04-13:
Thanks for the link, I was about to try the circle-stdlib fork with my build tools. I will give it a go and update the results.

ArnoldUK, 2023-04-13:
I've now managed to resolve the type conflicts by adding the following define at the top of kernel.h
 `#define _USE_LONG_TIME_T`

Appears that time_t defined as long in time.h was the problem. No need to download extra toolchains or circle-stdlib.

smuehlst, 2023-04-13:
Not sure what your goal is here: It won't work if you are trying to use stdio functions from a standard newlib library that comes from an ARM toolchain that comes from Debian itself. That is what circle-stdlib provides. The newlib built by circle-stdlib is based on Circle, so that for example stdio operations use Circle's I/O functions internally.

ArnoldUK, 2023-04-13:
I'm attempting to port this [mikechambers84/XTulator](https://github.com/mikechambers84/XTulator) to circle SDK. I've manged to compile and link all the c source files. Now working on porting all the SDL calls to circles screen framebuffer.

smuehlst, 2023-04-13:
Try to build a simple standalone program with your toolchain that reads and/or writes a file through stdio. You will see that it does not work.
If you want to use stdio with Circle, you must use it with a supported toolchain together with circle-stdlib. Just trying to save you some time here.

smuehlst, 2023-04-13:
> I'm porting all the stdio stuff to circle

Then you're basically reimplementing what is implemented in circle-stdlib already...

ArnoldUK, 2023-04-13:
I've not tried circle-stdlib and I think I should as you suggest and I guess I'm making hard work for myself doing this manually. Still, I'm learning how the SDK works as I go, not very keen on C language or CPP so much a learner in that respect.

rsta2, 2023-04-13:
Circle does not support using <stdio.h> standalone officially. The recommended way is to use circle-stdlib and the referenced toolchain for this purpose. Of course, you can implement your own solution here, if you like, but there is no support for this by us.

ArnoldUK, 2023-04-13:
Sure, I know that is the correct and preferred way but I'm only looking to get this partially running and would much liek to avoid downloading and installing another 15K+ files onto my flash disk for only a couple of stdio function calls. I'm sure I can use the current circle SDK for the file IO and timing functions. I've already had succes with an SDL interface to circle's framebuffer. Thanks for your valuable input.

rsta2, 2023-04-14:
OK, understood. Good luck.

ArnoldUK, 2023-04-14:
I'm closing this as not really an issue with the circle SDK but more a conflict with platform toolchain libraries and targets.

closed 2023-04-14 by ArnoldUK