rsta2/circle #365: Raspberry how to boot from USB (SSD) when an SD card is present?

mgnthe54, 2023-05-03:
Good morning,
I have a raspberry pi 4 running bullseye 64 bit.
I configured the boot on USB (SSD disk with external power supply to be sure that it is operational before restarting the PI) then if not available on SD:
sudi raspi-config ->option 6 Advanced Options
option A6 Boot order
B2 USB boot.
In the micro SD slot of the raspberry I have an SD containing a clone (via SD Card Copier) of my SSD disk.
If I reboot my Raspberry I start each time on the SD (unless the SD is not in the slot of course!)
How to boot from the SSD via USB when a micro SD is in the Pi?

THANKS

ArnoldUK, 2023-05-03:
There is a special bootcode.bin that can be placed onto the micro-sd card which allows booting from USB drives but I'm not sure if it would in your case. You can always try it anyway.
[special boot modes](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-boot-modes)
[bootcode.bin](https://github.com/raspberrypi/firmware/raw/master/boot/bootcode.bin)

mgnthe54, 2023-05-04:
Hello, thank you for your link,
I discovered that the raspberry Pi4 (mine therefore) use an EEPROM to select the boot:

"The Raspberry Pi 4B does not use the bootcode.bin file - instead the bootloader is located in an on-board EEPROM chip. See Raspberry Pi 4 Bootflow and SPI Boot EEPROM."

But it doesn't seem to work anyway with Bullseye, I'll have to redo an SD with Buster to test...

thanks again

rsta2, 2023-05-04:
The boot order can be configured for the RPi 4, with these [options](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER). Be careful, when modifying the boot configuration, so that your RPi 4 remains boot-able!

BTW. Please don't create multiple comments to Issues and Discussions with the same contents!

mgnthe54, 2023-05-04:
Hello again, I carried out other tests with Buster and Bullseye the problem is the same with the 2 systems, but I "slightly" progressed on the origin of the problem:
Although apparently the boot of the pi 4 is on an EEPROM of this one when the micro SD contains a clone of the SSD one boots on the micro SD and no longer on the USB!
Indeed I work with an SSD in USB but I leave a micro SD in the Raspberry to carry out every week a complete backup clone of the SSD on the SD via the command:
sudo rpi-clone mmcblk0 -v -U
And when the micro SD contains the contents of the SSD (clone) it is no longer seen on the Raspberry desktop and in the event of a reboot the pi starts on the micro SD and no longer on the USB.
Whereas if for example the SD contains a system (like Buster) but not a clone of the SSD (it with Bullseye or another Buster) it works well, I boot on the USB and on the desktop the 2 partitions of the micro SD appear well!
Here is where I am....
THANKS

rsta2, 2023-05-04:
I also think, there are better places to report/discuss this, because it's not specific to Circle:

* https://github.com/raspberrypi/firmware/issues
* https://forums.raspberrypi.com/

mgnthe54, 2023-05-04:
During my research I came across this:
pi@raspberrypi:~$ **sudo blkid**
**/dev/sda1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="6229-BAA7**" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="392a8b00-01"
/dev/sda2:LABEL="rootfs" UUID="7c7dfdc9-556c-4551-bb46-391b1dd27cc0" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="392a8b00-02"
/dev/sdc1:LABEL="BACKUPS" UUID="BC53-1C04" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="41505f42-01"
/dev/sr0: BLOCK_SIZE="2048" UUID="2013-11-21-16-45-01-00" LABEL="MegaFon" TYPE="iso9660" PTTYPE="mac"
**/dev/mmcblk0p1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="6229-BAA7**" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="3203184d-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="7c7dfdc9-556c-4551-bb46-391b1dd27cc0" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="3203184d-02"

The SD card (/dev/mmcblk0p1) the bootfs has the same UUID (6229-BAA7) as the ssd disk (/dev/sda1) whereas normally to my knowledge a UUID is supposed to be unique…

rsta2, 2023-05-04:
> I'm starting to think that it would come from another problem ...

Yes, and I'm missing any relation to Circle. At least you have not mentioned anything, which points in this direction. So this is not the right place to discuss this and I will close this now.

closed 2023-05-04 by rsta2