rsta2/circle #368: sample/04-timer doesn't work on qemu or real pi

MatthewRobertDunn, 2023-05-27:
I've tried to compile circle and sample04 with the recommended version of the arm tool chain, and I always get the same result:

`$ cat Config.mk
PREFIX = arm-none-eabi-
AARCH = 64
RASPPI = 3
DEFINE += -DNO_SDHOST
DEFINE += -DNO_CALIBRATE_DELAY
DEFINE += -DNO_PHYSICAL_COUNTER
`

$ aarch64-none-elf-g++.exe --version
aarch64-none-elf-g++.exe (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)) 12.2.1 20221205
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ qemu-system-aarch64 --version
QEMU emulator version 8.0.0
Copyright (c) 2003-2022 Fabrice Bellard and the QEMU Project developers

$ qemu-system-aarch64 -M raspi3b -kernel ./kernel8.img
`

[image]

And on the real rpi3 I get exactly the same result:

[image]

rsta2, 2023-05-28:
Unfortunately I cannot reproduce the problem. *sample/04-timer* is working here on a RPi 3B and in QEMU. I can only guess, that there were stale object files in your project, after modifying the configuration in *Config.mk*, which caused this problem. You have to clean the whole project after changing *Config.mk* with `makeall clean` and also the sample's subdirectory with `make clean`. Tag *Step45.2* is currently identical with the *master* branch.

MatthewRobertDunn, 2023-05-28:
Thanks for this. I must of corrupted my environment somehow I guess, or perhaps make clean is missing some files?

I'll try a fresh checkout of master

```
matthew@AMD UCRT64 ~/circle
$ git pull
remote: Enumerating objects: 12, done.
remote: Counting objects: 100% (12/12), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 12 (delta 7), reused 12 (delta 7), pack-reused 0
Unpacking objects: 100% (12/12), 2.11 KiB | 47.00 KiB/s, done.
From https://github.com/rsta2/circle
   24fa057..fd43595  develop    -> origin/develop
Already up to date.

matthew@AMD UCRT64 ~/circle
$ ./makeall clean
  CLEAN  /home/matthew/circle/tools
  CLEAN  /home/matthew/circle/lib
  CLEAN  /home/matthew/circle/lib/usb
  CLEAN  /home/matthew/circle/lib/input
  CLEAN  /home/matthew/circle/lib/fs
  CLEAN  /home/matthew/circle/lib/fs/fat
  CLEAN  /home/matthew/circle/lib/sched
  CLEAN  /home/matthew/circle/lib/net
  CLEAN  /home/matthew/circle/lib/sound
  CLEAN  /home/matthew/circle/sample/42-soundinput

matthew@AMD UCRT64 ~/circle
$ ./makeall -j
  COPY  converttool.exe
  CPP   actled.o
  CPP   alloc.o
  CPP   assert.o
  CPP   bcmframebuffer.o
  CPP   bcmmailbox.o
  CPP   bcmpropertytags.o
  CPP   bcmwatchdog.o
  CPP   chargenerator.o
  CPP   classallocator.o
  CPP   cputhrottle.o
  CPP   debug.o
  AS    delayloop.o
  CPP   device.o
  CPP   devicenameservice.o
  CPP   dmachannel.o
  CPP   gpioclock.o
  CPP   gpiomanager.o
  CPP   gpiopin.o
  CPP   gpiopinfiq.o
  CPP   i2cmaster.o
  CPP   i2cslave.o
  CPP   koptions.o
  CPP   logger.o
  CPP   machineinfo.o
  CPP   multicore.o
  CPP   nulldevice.o
  CPP   ptrarray.o
  CPP   ptrlist.o
  CPP   pwmoutput.o
  CPP   qemu.o
  CPP   screen.o
  CPP   serial.o
  CPP   spimaster.o
  CPP   spimasteraux.o
  CPP   spimasterdma.o
  CPP   spinlock.o
  CPP   string.o
  CPP   sysinit.o
  CPP   time.o
  CPP   timer.o
  CPP   tracer.o
  CPP   usertimer.o
  CPP   util.o
  AS    util_fast.o
  CPP   virtualgpiopin.o
  CPP   chainboot.o
  CPP   macaddress.o
  CPP   netdevice.o
  CPP   new.o
  CPP   heapallocator.o
  CPP   pageallocator.o
  AS    setjmp.o
  CPP   numberpool.o
  CPP   latencytester.o
  CPP   writebuffer.o
  CPP   2dgraphics.o
  CPP   smimaster.o
  CPP   ptrlistfiq.o
  CPP   exceptionhandler64.o
  AS    exceptionstub64.o
  CPP   memory64.o
  AS    startup64.o
  CPP   synchronize64.o
  CPP   translationtable64.o
  CPP   bcmrandom.o
  CPP   interrupt.o
  CPP   mphi.o
  CPP   purecall.o
  CPP   cxa_guard.o
  AR    libcircle.a
  CPP   lan7800.o
  CPP   smsc951x.o
  CPP   usbbluetooth.o
  CPP   usbcdcethernet.o
  CPP   usbconfigparser.o
  CPP   usbdevice.o
  CPP   usbdevicefactory.o
  CPP   usbendpoint.o
  CPP   usbfunction.o
  CPP   usbgamepad.o
  CPP   usbgamepadps3.o
  CPP   usbgamepadps4.o
  CPP   usbgamepadstandard.o
  CPP   usbgamepadswitchpro.o
  CPP   usbgamepadxbox360.o
  CPP   usbgamepadxboxone.o
  CPP   usbhiddevice.o
  CPP   usbhostcontroller.o
  CPP   usbkeyboard.o
  CPP   usbmassdevice.o
  CPP   usbmidi.o
  CPP   usbmouse.o
  CPP   usbprinter.o
  CPP   usbrequest.o
  CPP   usbstandardhub.o
  CPP   usbstring.o
  CPP   usbserial.o
  CPP   usbserialch341.o
  CPP   usbserialcp210x.o
  CPP   usbserialpl2303.o
  CPP   usbserialft231x.o
  CPP   usbserialcdc.o
  CPP   usbtouchscreen.o
  CPP   dwhcidevice.o
  CPP   dwhciframeschednper.o
  CPP   dwhciframeschednsplit.o
  CPP   dwhciframeschedper.o
  CPP   dwhciregister.o
  CPP   dwhcirootport.o
  CPP   dwhcixactqueue.o
  CPP   dwhcicompletionqueue.o
  CPP   dwhcixferstagedata.o
  CPP   dwhciframeschediso.o
  AR    libusb.a
  CPP   keyboardbehaviour.o
  CPP   keymap.o
  CPP   mouse.o
  CPP   touchscreen.o
  CPP   mousebehaviour.o
  CPP   rpitouchscreen.o
  CPP   console.o
  CPP   keyboardbuffer.o
  CPP   linediscipline.o
  AR    libinput.a
  CPP   partition.o
  CPP   partitionmanager.o
  AR    libfs.a
  CPP   fatfs.o
  CPP   fatcache.o
  CPP   fatinfo.o
  CPP   fat.o
  CPP   fatdir.o
  AR    libfatfs.a
  CPP   task.o
  CPP   scheduler.o
  AS    taskswitch.o
  CPP   synchronizationevent.o
  CPP   mutex.o
  CPP   semaphore.o
  AR    libsched.a
  CPP   netsubsystem.o
  CPP   nettask.o
  CPP   socket.o
  CPP   netsocket.o
  CPP   transportlayer.o
  CPP   networklayer.o
  CPP   linklayer.o
  CPP   netdevlayer.o
  CPP   phytask.o
  CPP   arphandler.o
  CPP   icmphandler.o
  CPP   routecache.o
  CPP   netconnection.o
  CPP   udpconnection.o
  CPP   tcpconnection.o
  CPP   retransmissionqueue.o
  CPP   retranstimeoutcalc.o
  CPP   tcprejector.o
  CPP   netconfig.o
  CPP   ipaddress.o
  CPP   netqueue.o
  CPP   checksumcalculator.o
  CPP   dnsclient.o
  CPP   ntpclient.o
  CPP   mqttclient.o
  CPP   mqttsendpacket.o
  CPP   mqttreceivepacket.o
  CPP   dhcpclient.o
  CPP   ntpdaemon.o
  CPP   httpdaemon.o
  CPP   httpclient.o
  CPP   tftpdaemon.o
  CPP   syslogdaemon.o
  AR    libnet.a
  CPP   dmasoundbuffers.o
  CPP   hdmisoundbasedevice.o
  CPP   pwmsoundbasedevice.o
  CPP   i2ssoundbasedevice.o
  CPP   pwmsounddevice.o
  CPP   soundbasedevice.o
  CPP   pcm512xsoundcontroller.o
  CPP   wm8960soundcontroller.o
  AR    libsound.a
  CPP   main.o
  CPP   kernel.o
  LD    kernel8.elf
  DUMP  kernel8.lst
  COPY  kernel8.img
  WC    kernel8.img => 101944

matthew@AMD UCRT64 ~/circle
$ cd sample/

matthew@AMD UCRT64 ~/circle/sample
$ cd 04-timer/

matthew@AMD UCRT64 ~/circle/sample/04-timer
$ make clean
  CLEAN  /home/matthew/circle/sample/04-timer

matthew@AMD UCRT64 ~/circle/sample/04-timer
$ make
  CPP   main.o
  CPP   kernel.o
  LD    kernel8.elf
  DUMP  kernel8.lst
  COPY  kernel8.img
  WC    kernel8.img => 72152

matthew@AMD UCRT64 ~/circle/sample/04-timer
$ qemu-system-aarch64 -M raspi3b -kernel ./kernel8.img

matthew@AMD UCRT64 ~/circle/sample/04-timer
$ qemu-system-aarch64 -M raspi3b -kernel ./kernel8.img
```

Result: 

[image]

MatthewRobertDunn, 2023-05-28:
I did a clean checkout of the development branch and it works entirely fine, so I'm not sure how I polluted my original folder. 

Sorry for the trouble :(

closed 2023-05-28 by MatthewRobertDunn