rsta2/circle #377: Application specific cmdline.txt options

toptensoftware, 2023-05-29:
Is there a way for an circle application to have its own custom cmdline.txt parameters?

Use case:  I've got an app that runs on multiple devices, but doesn't need SD card access.  Each needs to be configured slightly differently (basically a different I2C address for each one) and cmdline.txt would be an easy way to configure this.

What if `CKernelOptions` kept track of anything it didn't recognize as a separate string so the app could get anything left over?

(again I'm happy to implement this after hearing your thoughts)

rsta2, 2023-05-29:
I will provide a solution for this, as suggested. `CKernelOptions` will save all values for options, it does not know and offer the following new methods:

```
const char *GetAppOptionString (const char *pOptionName);
unsigned GetAppOptionDecimal (const char *pOptionName);
```

rsta2, 2023-05-29:
Done.

toptensoftware, 2023-05-29:
Awesome - thanks!

closed 2023-05-29 by toptensoftware