rsta2/circle #384: Raspi 4 doesn't like kernel8-32.img

toptensoftware, 2023-06-29:
I've been trying to chase down an issue and narrowed it down to `kernel8-32.img` on Raspi 4.

I've reproduced the same thing with the `01-simplegpio` sample as follows:

1. Build all 6x configs of the project - raspi 1,2,3,4 aarch32 and raspi 3,4 aarch 64. (or [get them here](https://www.dropbox.com/home/Public/01-gpiosimple))
2. Put all 6 images on a SD card, with other boot files and set `config.txt` set for aarch32.
3. Boot it on a Raspberry Pi 4.
4. Note: no flashing LED.
5. Now delete `kernel8-32.img` from the SD card
6. Boot again and now LED flashes.

If `kernel8-32.img` is for rpi3 and `kernel7l.img` is for rpi4 why does it look like rpi4 is trying to load `kernel8-32.img`?

What's going on here - am I misunderstanding something?

toptensoftware, 2023-06-29:
I managed to work around this by adding a `kernel` key to the `[pi4]` section in `config.txt`.  

But I'm still curious whether this is expected and/or should be required?

```
[pi4]
armstub=armstub7-rpi4.bin
max_framebuffers=2
kernel=kernel7l.img
```

rsta2, 2023-06-29:
The only official documentation about the naming of kernel images, that I found, is [here](https://www.raspberrypi.com/documentation/computers/config_txt.html#kernel). *kernel8-32.img* is not mentioned here. Personally I normally prefer to use *kernel7.img*, which runs on both RPi 2 and 3, but is optimized for RPi 2. It's good, if you found a way for naming kernel images for your purpose. Unfortunately I cannot tell you much more about this.

toptensoftware, 2023-06-29:
Thanks Rene.  Now that I know it's not a big deal, but I spent a long time tracking this down. It was just confusing because I was building all targets and putting them on the card, thought my program was broken, but really it was pi4 preferring the pi3 image - which obviously didn't work.  It was only when I deleted the image I wasn't working on that things started working again.

closed 2023-06-30 by toptensoftware

rsta2, 2023-06-30:
I understand, but this is a firmware thing. Nevertheless I will add an info about *kernel8-32.img* to the main README.

toptensoftware, 2023-06-30:
Thanks Rene, yep understand this is firmware related (in fact I tried the latest firmware before opening this issue).  I only opened the issue as I thought you might have some insight.  A note mentioning this is a good idea