rsta2/circle #386: Kernel image size increase

dlbogdan, 2023-06-29:
Hi,
I want to embed fonts and other glyps and graphics in the source thus bloating the kernel image beyond 2MB potentially quite quickly. 
is there a way to increase the 2MB limit for the kernel image size? 

Reference:
doc/memorymap.txt

dlbogdan, 2023-06-29:
Thanks! I wasn’t aware of sysconfig.h at all.

closed 2023-06-29 by dlbogdan

rsta2, 2023-06-29:
It's mentioned in the main README. ;)

dlbogdan, 2023-06-29:
Oops :)

ArnoldUK, 2023-06-29:
Yes, the `sysconfig.h` was recently updated to allow passing `KERNEL_MAX_SIZE` in your makefile without having to change the value in `sysconfig.h` so multiple projects can have a different size kernel.
Do Remember to rebuild the circle libs after changing it as I found out after hours of hair pulling.

dlbogdan, 2023-06-29:
Oh I rebuild circle, stdlib/newlib lvgl and a few others pretty often (at least once a day) as I test my project with a pi0 and a pi3A+ and I am very interested in performance so it matters to recompile.
Thanks for the tip!